micro-ROS/micro_ros_espidf_component #172: Adding custom library of sensors with micro-ROS freertos

CasCard, 2023-01-03:
## Issue template

- Hardware description: ESP32-WROOM32
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

I am having Adafruit VL6180X ToF sensor, but it uses a [custom library with Adafruit headers](https://github.com/adafruit/Adafruit_VL6180X). I would like to know how can I include this library when creating the FreeRTOS micro-ros application. 
My application is to publish the sensor data from ESP32 using micro-ros , which can be subscribed by raspberry pi with ROS2 running.

pablogs9, 2023-01-04:
Hello @CasCard, the folder structure you show is not generated by `micro_ros_espidf_component`. Could you please use `micro_ros_espidf_component`, so the integration of your library will be as usual in ESP-IDF.

CasCard, 2023-01-23:
I was finally able to solve this issue. If someone looking to integrate Arduino libraries into your project here are the steps to follow.

- Install ESP-IDF Version using VS Code (I installed ESP IDF version 4.4 Version)
- Now we need to create a sample ESP-IDF project, you can get this template from `View`->`ESP-IDF examples`->`sample project`
- A typical project file would consist of the following folder (or structure), (some files are auto-generated)

[image]
- Now the `components` folder is where the additional libraries including `micro_ros_espidf_component` reside.

If you want to specifically follow the format usually followed for Arduino includes `setup` and `loop` format you have to include a library called [arduino-esp32](https://github.com/espressif/arduino-esp32) in the components folder as shown in the above image.

Now if you want to include Arduino libraries you can directly clone them into the `components` folder. Here is an example where I am including the [Adafruilt VL6180X library](https://github.com/adafruit/Adafruit_VL6180X).

[image]

Now you have to additionally add `CMakeLists.txt` in the library so that IDF will be able to find them while building.You can keep the header files in the `include` folder. You might need to add additional supporting libraries for your core libraries.
For example Adafruilt VL6180X requires [arduino-esp32](https://github.com/espressif/arduino-esp32) and [Adafruit_BusIO](https://github.com/adafruit/Adafruit_BusIO) libraries.

So the CMakeLists.txt of Adafruilt VL6180X library should look like as follows:

`
idf_component_register(SRCS "Adafruit_VL6180X.cpp"
                       INCLUDE_DIRS "include"
                       REQUIRES arduino Adafruit_BusIO)
`

Now for writing arduino ide like code you can follow [this](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html#:~:text=Latest%20Arduino%20Core%20ESP32%20version,compatible%20with%20ESP%2DIDF%20v4) reference.

Hope this helps.

closed 2023-01-23 by CasCard