micro-ROS/micro_ros_espidf_component #173: custom transport with serial devices without agent

ManvithRaj, 2023-01-05:
Hi @pablogs9 ,

We have a requirement for micro-ros with serial devices connected in a loop. 

So we are looking for micro-ros solution with Serial UART Communication without agent.

Could you please let me know is it possible if yes please let me know the steps to achieve it.

Thanks in advance.

pablogs9, 2023-01-05:
Hello @ManvithRaj, now micro-ROS does not support communication without an Agent using serial devices. 

In any case, we can provide services using other (non open sourced) eProsima products that may fit your requirements. Please contact our sales department: info@eProsima.com

closed 2023-01-05 by pablogs9

ManvithRaj, 2023-01-05:
Hi @pablogs9 

Thank you so much for the clarification

ManvithRaj, 2023-01-05:
Hi @pablogs9 

Could you please reply to below another issue created last week. i am waiting for your reply

https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/339