micro-ROS/micro_ros_espidf_component #174: Parameters issue

ALUIS97, 2023-01-29:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: ESP IDF VS CODE extension with the esp idf v4.4, micro ros espidf component in the folder components of my project, Micro-Ros agent from micro_ros_setup
- Version or commit hash: Humble (default)

#### Steps to reproduce the issue

1. Copy and paste the example parameters of this repository into the main of the project created for vs code esp idf extension
2. change the following line of code highlighted in blue to a value of 10 because I didn't find the RCLC_PARAMETER_EXECUTOR_HANDLES_NUMBER 
[image]

1. change colcon.meta to services 10 and rmw_microxrcedds/config.h service to 10
2. set wifi port and agent ip correctly 
3. build and flash
4. execute ros agent udp port 8888 
5. execute the ros2 param list

#### Expected behavior
To see param1,2 and 3 in the list 

#### Actual behavior
I don't see my parameters on the list after executing the param list command or param get  node_name param1
[image]

But from the ros agent terminal, I see the ros agent is sending a message to the micro ros node but I don't have any response and from the topic created for this example we get 
[image]

[parameter_verbose6.txt.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/10530551/parameter_verbose6.txt.txt)

#### Additional information

I have increased my heap memory to 60 kbytes

I tried to follow the next issue 
https://github.com/micro-ROS/micro_ros_zephyr_module/issues/34
I tried to change my docker file 
[image]

But I don't know how to change the ros agent of the micro_ros_Setup repo.

I don't know if DDS is not allowing to receive things from parameters

Acuadros95, 2023-01-30:
> change colcon.meta to services 10 and rmw_microxrcedds/config.h service to 10

Did you rebuild your micro-ROS library after this modification? 
`rmw_microxrcedds/config.h` should not be modified, please check that the generated `config.h` contains the expected number of services without manual modifications.

Acuadros95, 2023-01-30:
From the README: 

To clean and rebuild all the micro-ROS library:
```bash
idf.py clean-microros
```

Anyway, We should update the repo documentation, as its not clearly stated or documented.

ALUIS97, 2023-01-30:
thanks well yes I agree with you I think there are a lot of things missing related to documentation, anyway with vs code extension I see libmicroros.a change
[image]

ALUIS97, 2023-01-30:
thank youuuu very much is working right now

closed 2023-01-30 by ALUIS97