micro-ROS/micro_ros_espidf_component #179: Example Code on how to use GPIO Interrupt with MicroRos

GlingK, 2023-02-24:
I'm trying to have micro ros publish sensor information when an ESP32 pin is interrupt. 
So far, 
1. Micro ros with interrupts, the ESP32 reboots due to panic_abort.
2. Micro ros alone, using timers to read the pin it works fine.
3. Interrupts alone also works fine

How would one go about using micro-ros with interrupts? I've try searching around, but didn't find any sample code. Any guidance is much appreciated.

pablogs9, 2023-02-24:
Probably you will use FreeRTOS to notify a task from the IRQ to publish from a task context. 
- I'm not sure, but micro-ROS is quite heavy to be called from the context (stack) of an IRQ.
- Also, normally is not a good idea to call a long or blocking method in an IRQ, and micro-ROS publish method can take a bit or even block (depending on the configuration).
- Also, in the IRQ context probably you are not allowed to call the complete system API, normally it is limited to a small set of IRQ-safe APIs.

Let us know if notifying a task from an IRQ is a valid approach for you: https://www.freertos.org/xTaskNotifyFromISR.html

asasine, 2023-05-05:
I've tried to implement a simple interrupt -> publish application and I've encountered issues making it difficult. What I originally tried is the same as what @pablogs9 recommended: have an ISR notify a suspended task with `xTaskNotifyFromISR`, and have that task publish a ROS message. Because that suspended task is not the same task as the micro-ROS task, it's unsafe to publish a message from it without enabling the `UCLIENT_PROFILE_MULTITHREAD` build flag. However, as reported in #155, enabling that flag is not possible.

I've resorted to running an rclc timer that publishes the message periodically, and setting the fields in the message from the ISR, however this either adds a delay to when the message is published because of the timer, or wastes processing on a faster timer to get the necessary reactivity.