micro-ROS/micro_ros_espidf_component #181: ESP32 requires reset every time agent is launched

CarlHVA, 2023-03-30:
## Issue template

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: ESP IDF VS CODE extension with the esp idf v4.4, micro ros espidf component in the folder components of my project, Micro-Ros agent from micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue
Running the micro-ros agent.

docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev [YOUR BOARD PORT] -v6

#### Expected behavior
The ESP32 starts running the micro ros code after launching the agent.

#### Actual behavior
The code runs only after pressing the reset button on the ESP32 board. Until then, the only output I get is this:
[image]

#### Additional information
I'm using a serial connection. The code runs fine after resetting the board. However, it is a hassle needing to reset the board each time I want to launch the agent.

pablogs9, 2023-03-30:
You can implement a reconnection approach: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html

CarlHVA, 2023-03-30:
Thanks for the very quick reply and helpful answer! This solved my issue.