micro-ROS/micro_ros_espidf_component #183: libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it

Usama-Arshad16, 2023-04-03:
## int32_publisher example not building

- Hardware description: I was trying to build it for esp32
- Installation type: ROS2 foxy with micro_ros_espidf_component
- Version or commit hash: microros foxy

#### Steps to reproduce the issue
I have Ros Foxy installed on my PC. In order to use microros with esp32, I first cloned esp_idf and then I cloned micro_ros_espidf_component repo. 
These are the commands i followed

- . $IDF_PATH/export.sh
- cd examples/int32_publisher
- idf.py set-target esp32
- idf.py build

#### Expected behavior
I was expecting it to build successfully.
#### Actual behavior
But Actually it was giving me error shown below

[image]
[image]

#### Additional information

liamhan1, 2023-04-04:
I'm getting the same error

nilseuropa, 2023-04-05:
Me too,but  on humble branch.

UsamaArshad16, 2023-04-12:
Please someone solve this.
I also tried this using docker, but the error stays the same
[image]

glmnet, 2023-06-05:
I was getting this error then I run 
```
cd ~/esp/esp-idf
source ./export.sh
pip3 install catkin_pkg lark-parser empy colcon-common-extensions
```

then I clean build and it worked, seems like the source before pip3 install didn't work when I did it first time.

dgarcu, 2023-06-19:
I have also solved it with the solution provided by @glmnet :smile:

closed 2025-07-15 by 4ntn