micro-ROS/micro_ros_espidf_component #186: Int32_publisher build problems

Apurva2510, 2023-04-12:
- Hardware description: esp32, esp32s3
- RTOS: freertos
- Installation type: 
- Version or commit hash: humble with esp-idf v5.1, Ubuntu 22.04 LTS

#### Steps to reproduce the issue
1. Open esp-idf as an extension on VS code
2. Get the int32_publisher example from the micro_ros_espidf_component repository
3. open the int32_publisher example in the VS code esp-idf extension
4. Select the board as esp32/esp32s3 from the set target device on the VS code by selecting the int32_publisher as the folder and the board as espe32 or esp32s3 and finally select ESP32 chip (via ESP USB BRIDGE). The board will be selected.
5. Click on build

#### Actual behavior

[image]

[image]

#### Additional information

pablogs9, 2023-04-12:
Are you able to build a project without micro-ROS? I do not see any micro-ROS related error on your logs.

Apurva2510, 2023-04-12:
> Are you able to build a project without micro-ROS? I do not see any micro-ROS related error on your logs.

yes I have tried without micro-ROS and it works. Also when I try to build the project from the terminal from the micro_ros workspace it seems to build and flash as well on the esp32 board.

The above issue only happens when I try to build and flash through the esp-idf VS code extension

pablogs9, 2023-04-12:
I guess that this extension is not well supported. Feel free to contribute with a PR if you find a solution

Apurva2510, 2023-04-12:
> I guess that this extension is not well supported. Feel free to contribute with a PR if you find a solution

[image]

Also these libraries don't seem to be recognized. Can this be an issue related to the PATH of the libraries?

4ntn, 2025-09-29:
Hello, @Waelhechmi.

The `microros/esp-idf-microros` is deprecated and has not been updated in a long time. The recommended way to use `micro_ros_espidf_component` is with the official espressif Docker container and installing the micro-ROS dependencies manually. The [Readme](https://github.com/micro-ROS/micro_ros_espidf_component?tab=readme-ov-file#build-with-docker-container) has been updated to reflect this.

Another possibility is to build the Docker image locally using the `docker/Dockerfile` in this repo to have an up-to-date image.