micro-ROS/micro_ros_espidf_component #187: rcl/rcl.h: No such file or directory

tianrking, 2023-04-14:
https://github.com/tianrking/ESP32_MicroROS

At one time I was able to compile in the old version but the current version gives this error. I'm sure the environment is clean, I've enabled the espidf environment and configured the environment variables, but I can't find rcl rcl.h

[100%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/root/ESP32_MicroROS/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:76: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:4839: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
make failed with exit code 2

Digitelektro, 2023-04-27:
A ran into the same problem. I figured out the problem (at least in my case).
 - Do not source ROS2 as stated in the readme!
 - If you had tried to build it with ROS2 sourced, make sure you remove the build folder and also run `idf.py clean-microros`
 - For me building libmicroros is failed with idf-4.2, I installed a fresh idf-4.4.4 and then the required python packages in a terminal where idf is sourced.
 - With the correct idf libmicroros build should be successful as well as the project. If not, check build output for details.

MJohn2, 2023-09-01:
> Could you provide the whole log on an clean environment?

Hi Pablo
I've made the mistake of building with Ros 2 sourced by inserting it into bashrc and forgetting about it. 
I've followed @Digitelektro's steps above but it's failing to build as per this original issue.
Can you point me to some information on what a clean environment looks like?
TIA

flabrosse, 2023-11-27:
Trying to build example/int32_publisher following the instructions in README.md.  I have not sourced any of the ROS2 setup files.  esp-idf-4.4.6.  From a fresh clone, I get:

```
[ 99%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/home/ffl/local/src/micro_ros_espidf_component/examples/int32_publisher/main/main.c:11:10: fatal error: rcl/rcl.h: No such file or directory
 #include <rcl/rcl.h>
          ^~~~~~~~~~~
```

flabrosse, 2023-12-25:
Trying any of the branches of `micro_ros_espidf_component`, the whole of `micro_ros_src` folder is missing, ie is not pulled in by the build process, and therefore `rcl.h` (and others) is missing.  If I use the component [here](https://github.com/Aber-CRANC/micro_ros_espidf_component) then it all works fine.  Apart from the obvious branch differences, I cannot see what makes one fail while the other works.  It looks like the target `$(EXTENSIONS_DIR)/micro_ros_src/src` in `libmicroros.mk` is not executed, but so far I have not been able to figure out why.

flabrosse, 2024-01-04:
The reason the `micro-ros-src` (and other) folder is not pulled in is because the `execute_process` function in the `CMakeLists.txt` file fails.  It fails because `submake` expands to `$(MAKE)` on Unix which either is not expanded further or not defined.  If I set `submake` to `make` (or replace `${submake}` with `make` in the `execute_process` command, then all the necessary files are pulled in.

I am not entirely sure why this is the case, but this works.  Should `MAKE` be an environment variable?  Can't `submake` be set to `make` all the time?

flabrosse, 2024-01-11:
@pablogs9 would you have views on this?  I am happy to prepare a PR but I am not sure that just setting `submake` to `make` in all cases is the right approach.

I have tried to set `MAKE` in my environment, or to call `idf.py build -DMAKE=make`, to no avail.

pablogs9, 2024-01-11:
Please open a PR so we can discuss there about your proposed changes

flabrosse, 2024-01-11:
Done (PR #223).  I have worked on the humble branch as this is the one I use.

Waelhechmi, 2025-02-27:
hi guys , i will to create a new project not build a example how i do it !

fnndyl, 2025-05-15:
Hi @flabrosse and @pablogs9, I seem to still be getting this issue with the changed CMakeLists.txt file - I will try it in a ROS2 ubuntu docker (and on a macbook) but on my Ubuntu 22.04 desktop getting this issue trying to compile the int32 example. Any ideas?

fnndyl, 2025-05-15:
Same problem on MacOS as well hm. Trying to build int32_publisher.

flabrosse, 2025-05-15:
I have not updated my version of the code in a while, so I am not sure.  Do you get the same error message?

Waelhechmi, 2025-05-16:
use the docker container of that repo .