micro-ROS/micro_ros_espidf_component #188: esp-wroom-32 rebooting after flashing the multithread_publisher in it with udp as transport

Apurva2510, 2023-04-20:
## Issue template

- Hardware description: ESP-WROOM-32 devkit
- RTOS: freertos
- Installation type: 
- Version or commit hash: Humble

#### Steps to reproduce the issue
build the multithread_publisher and flash it on the micro controller the standard way and check the idf monitor.

#### Expected behavior
ESP-WROOM-32 not rebooting

#### Actual behavior
Agent
[image]

Client
[image]

#### Additional information

ALUIS97, 2023-07-03:
i have the same problem

demon4eg, 2023-12-03:
So am i