micro-ROS/micro_ros_espidf_component #189: Not able to build code int32_publisher_custom_transport over UART

nilutpolkashyap, 2023-04-23:
- Hardware description: ESP32 Dev Board 3.3V
- RTOS: FreeRTOS
- Installation type: esp-idf component
- Version or commit hash: humble

#### Steps to reproduce the issue
- Copied the code from 'int32_publisher_custom_transport/main/main.c' into 'esp32_ros2_example/main/main.c'
- Copied 'Kconfig.projbuild' file inside 'esp32_ros2_example/main' directory
- Copied 'esp32_serial_transport.h' and 'esp32_serial_transport.c' files inside 'esp32_ros2_example/main' directory
- Updated 'main/CMakeLists.txt' to include 'esp32_serial_transport.c'
- Copied 'app-colcon.meta' and 'colcon.meta' files from 'int32_publisher_custom_transport' inside 'esp32_ros2_example' directory
- **'idf.py menuconfig'** and then changed 'micro-ROS network interface select' to 'Micro XRCE-DDS over UART' option inside 'micro-ROS settings'

### Folder structure
```
├── CMakeLists.txt
├── components
|   └── micro_ros_espidf_component
├── main
│   ├── CMakeLists.txt
    ├── component.mk
    ├── esp32_serial_transport.c
    ├── esp32_serial_transport.h
    ├── Kconfig.projbuild
│   └── main.c
├── app-colcon.meta
├── colcon.meta
└── README.md                  
```

#### Expected behavior
Building the project:
```
idf.py build
```
Successful build of the project in esp-idf.

#### Actual behavior
Building the project:
```
idf.py build
```
Not able to build the example code from [int32_publisher_custom_transport](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_publisher_custom_transport)
[image]

#### Additional information
Detailed build error log from 'idf.py build' command
```
Executing action: all (aliases: build)
Running cmake in directory /home/nilutpolk/esp/esp32_ros2_example/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/nilutpolk/esp/esp32_ros2_example"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nilutpolk/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/nilutpolk/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/nilutpolk/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/nilutpolk/esp/esp32_ros2_example/sdkconfig
-- Found Python3: /home/nilutpolk/.espressif/python_env/idf4.4_py3.10_env/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
[... 224 lines cut ...]
 bool esp32_serial_close(struct uxrCustomTransport * transport);
                                ^~~~~~~~~~~~~~~~~~
/home/nilutpolk/esp/esp32_ros2_example/main/esp32_serial_transport.h:25:34: warning: 'struct uxrCustomTransport' declared inside parameter list will not be visible outside of this definition or declaration
 size_t esp32_serial_write(struct uxrCustomTransport* transport, const uint8_t * buf, size_t len, uint8_t * err);
                                  ^~~~~~~~~~~~~~~~~~
/home/nilutpolk/esp/esp32_ros2_example/main/esp32_serial_transport.h:26:33: warning: 'struct uxrCustomTransport' declared inside parameter list will not be visible outside of this definition or declaration
 size_t esp32_serial_read(struct uxrCustomTransport* transport, uint8_t* buf, size_t len, int timeout, uint8_t* err);
                                 ^~~~~~~~~~~~~~~~~~
/home/nilutpolk/esp/esp32_ros2_example/main/main.c: In function 'app_main':
/home/nilutpolk/esp/esp32_ros2_example/main/main.c:97:2: error: #error micro-ROS transports misconfigured
 #error micro-ROS transports misconfigured
  ^~~~~
At top level:
/home/nilutpolk/esp/esp32_ros2_example/main/main.c:83:15: warning: 'uart_port' defined but not used [-Wunused-variable]
 static size_t uart_port = UART_NUM_0;
               ^~~~~~~~~
[1048/1055] Building C object esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_core/ed25519/ref10/ed25519_ref10.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

My custom package created in ESP-IDF is in this repository [https://github.com/nilutpolkashyap/esp32_ros2_example](https://github.com/nilutpolkashyap/esp32_ros2_example)

I have mentioned the whole process in the [README.md](https://github.com/nilutpolkashyap/esp32_ros2_example/blob/main/README.md) file of my [repository](https://github.com/nilutpolkashyap/esp32_ros2_example)

closed 2023-04-25 by nilutpolkashyap

nilutpolkashyap, 2023-04-26:
I solved it by running **'idf.py clean-microros'** inside the project directory to clean and rebuild all the micro-ROS library. Then, I was able to build and flash the project.

Waelhechmi, 2025-03-17:
Agent is not worked , if ros2 topic list the topic is not found