micro-ROS/micro_ros_espidf_component #190: ESP32 does not publish any topic over UART

nilutpolkashyap, 2023-04-26:
- Hardware description: ESP32 Dev Board 3.3V
- RTOS: FreeRTOS
- Installation type: esp-idf component
- Version or commit hash: humble, esp-idf  v4.4

My whole project is inside the repo https://github.com/nilutpolkashyap/esp32_ros2_example

#### Steps to reproduce the issue
- Make changes to 'colcon.meta' file inside 'components/micro_ros_espidf_component'
```
cd components/micro_ros_espidf_component/
sed -i 's/DRMW_UXRCE_TRANSPORT=udp/DRMW_UXRCE_TRANSPORT=custom/' colcon.meta
```
- Type **'idf.py clean-microros'** inside the project to clean and rebuild all the micro-ROS library
- Copied the code from 'int32_publisher_custom_transport/main/main.c' into 'esp32_ros2_example/main/main.c'
- Copied 'Kconfig.projbuild' file inside 'esp32_ros2_example/main' directory
- Copied 'esp32_serial_transport.h' and 'esp32_serial_transport.c' files inside 'esp32_ros2_example/main' directory
- Updated 'main/CMakeLists.txt' to include 'esp32_serial_transport.c'
- Copied 'app-colcon.meta' and 'colcon.meta' files from 'int32_publisher_custom_transport' inside 'esp32_ros2_example' directory
- Type **'idf.py menuconfig'** and then changed 'micro-ROS network interface select' to 'Micro XRCE-DDS over UART' option inside 'micro-ROS settings'
- In **idf.py menuconfig**, changed both **UART TX pin** and **UART RX pin** to **'0'** inside 'micro-ROS settings -> UART settings'

### Folder structure
```
├── CMakeLists.txt
├── components
|   └── micro_ros_espidf_component
├── main
│   ├── CMakeLists.txt
    ├── component.mk
    ├── esp32_serial_transport.c
    ├── esp32_serial_transport.h
    ├── Kconfig.projbuild
│   └── main.c
├── app-colcon.meta
└── README.md                  
```
I used the commands to build my project and then flash onto my ESP32 dev board
```
idf.py build
idf.py flash -p /dev/ttyUSB0
```
My ESP32 dev board com port is /dev/ttyUSB0

#### Expected behavior
Able to see data published by EPS32 by running micro-ROS agent docker.

#### Actual behavior
Running the micro-ROS agent docker 
```
docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyUSB0 -v6
```
gives the error
```
[1682516131.639975] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyUSB0, error 2, waiting for connection...
```
[image]

I tried to run the micro-ROS agent built inside my workspace,  
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 -v6
```
I got the following output
```
[1682517078.353975] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1682517078.354132] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

```
After that, I needed to press the Reset 'EN' button on my ESP32 board to see the following output. I get the **session established** output around 10 times, after which nothing is printed. 
```
[1682517113.228130] info     | Root.cpp           | create_client            | create                 | client_key: 0x19B62B22, session_id: 0x81
[1682517113.228291] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x19B62B22, address: 0
[1682517113.228560] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517114.230784] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517114.231030] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517114.231283] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517115.232003] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517115.232207] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517115.232520] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517116.232588] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517116.232749] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517116.232936] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517117.233720] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517117.233904] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517117.234086] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517118.237543] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517118.237717] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517118.237880] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517119.249222] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517119.249358] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517119.249555] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517120.249677] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517120.249831] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517120.250027] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517121.249633] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517121.249842] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517121.249992] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1682517122.249631] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x19B62B22, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 19 B6 2B 22 81 00 FC 01
[1682517122.249806] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x19B62B22, address: 0
[1682517122.249977] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x19B62B22, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00

```
[image]

Even running 'ros2 topic list' does not show any micro-ROS topics

#### Additional information

nilutpolkashyap, 2023-04-28:
@Acuadros95 any help would be highly appreciated.

Acuadros95, 2023-04-28:
> In idf.py menuconfig, changed both UART TX pin and UART RX pin to '0' inside 'micro-ROS settings -> UART settings'

Why? Could you give more details on your transport?

From the Agent log, it looks like the ESP is sending the data correctly, but is unable to retrieve the response from the agent.
Try to debug and fix the RX side of the ESP32 Serial transport

nilutpolkashyap, 2023-04-28:
> Why? Could you give more details on your transport?

I read through some previous issues which required changing the UART pins to 0. This step may be wrong. But either way, it didn't work for me.

> Try to debug and fix the RX side of the ESP32 Serial transport

Can you be more specific about what steps I need to take? I am not sure what to do here.

closed 2023-05-17 by nilutpolkashyap

Waelhechmi, 2025-03-17:
How to show topic , if tap ros2 topic list , not topic found