micro-ROS/micro_ros_espidf_component #191: int32_publisher not building for esp32c3

Apurva2510, 2023-05-05:
- Hardware description: ESP32C3-Wroom-02
- RTOS: freertos
- Installation type: Standard
- Version or commit hash: Humble

#### Steps to reproduce the issue
In the terminal:
1)Select int32_publisher example
2) idf.py set-target esp32c3
3)idf.py menuconfig (I used Ethernet as Transport)
4)idf.py build

#### Expected behavior
100% successful build

#### Actual behavior
[image]

[image]

#### Additional information

udit2720001, 2023-10-24:
same problem

Scoeerg, 2023-11-03:
@Apurva2510 Please write log-files, or better copy-paste into 

```
code blocks
```

for others to 

1) find the problem via Google. Google indexes text not images.
2) to read.