micro-ROS/micro_ros_espidf_component #194: Are services supported with EmbeddedRTPS?

jbmouret, 2023-05-24:
## Issue template

- Hardware description: M5Core2
- RTOS: FreeRTOS
- Installation type: EmbeddedRTPS
- Version or commit hash: 

#### Steps to reproduce the issue
- I tried to compile the add_two_ints example

#### Expected behavior
- compilation

#### Actual behavior
- I get a link issue:
```
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/project/components/micro_ros_espidf_component/libmicroros.a(librcl-service.c.obj):(.literal.rcl_service_init+0x0): undefined reference to `rmw_service_request_subscription_get_actual_qos'
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/project/components/micro_ros_espidf_component/libmicroros.a(librcl-service.c.obj):(.literal.rcl_service_init+0x4): undefined reference to `rmw_service_response_publisher_get_actual_qos'
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/project/components/micro_ros_espidf_component/libmicroros.a(librcl-service.c.obj): in function `rcl_service_init':
service.c:(.text.rcl_service_init+0xcf): undefined reference to `rmw_service_request_subscription_get_actual_qos'
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: service.c:(.text.rcl_service_init+0xe6): undefined reference to `rmw_service_response_publisher_get_actual_qos'
collect2: error: ld returned 1 exit status
```

And indeed, `rmw_service_response_publisher_get_actual_qos` seems to be defined nowhere.

#### Additional information

pablogs9, 2023-05-25:
Hello @jbmouret, services are not supported in Embedded RTPS for micro-ROS

jbmouret, 2023-05-25:
Thanks! so the error is expected.

closed 2023-05-25 by jbmouret