micro-ROS/micro_ros_espidf_component #199: multithread issue publisher

ALUIS97, 2023-07-04:
## Issue template

- Hardware description: 
- esp32 wroom
- RTOS: 
- freertos
- Installation type: 
- 
- Version or commit hash: 
- humble

#### Steps to reproduce the issue

try to build the micro ros component with multithread
#### Expected behavior

#### Actual behavior
/home/luis/esp/esp-idf/components/newlib/platform_include/errno.h:10:2: warning: #include_next is a GCC extension
   10 | #include_next "errno.h"
      |  ^~~~~~~~~~~~
---
Aborted  <<< rosidl_adapter [5.86s]

Summary: 4 packages finished [16.2s]
  1 package failed: microxrcedds_client
  2 packages aborted: rmw_implementation_cmake rosidl_adapter
  3 packages had stderr output: microxrcedds_client rcutils rosidl_cli
  67 packages not processed
make: *** [libmicroros.mk:90: /home/luis/Documents/PORYECT/uROS_mobile_robot_1/components/micro_ros_espidf_component/micro_ros_src/install] Error 2
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__bdc_motor espressif__mpu6050 esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls micro_ros_espidf_component mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/luis/esp/esp-idf/components/app_trace /home/luis/esp/esp-idf/components/app_update /home/luis/esp/esp-idf/components/bootloader /home/luis/esp/esp-idf/components/bootloader_support /home/luis/esp/esp-idf/components/bt /home/luis/esp/esp-idf/components/cmock /home/luis/esp/esp-idf/components/console /home/luis/esp/esp-idf/components/cxx /home/luis/esp/esp-idf/components/driver /home/luis/esp/esp-idf/components/efuse /home/luis/esp/esp-idf/components/esp-tls /home/luis/esp/esp-idf/components/esp_adc /home/luis/esp/esp-idf/components/esp_app_format /home/luis/esp/esp-idf/components/esp_common /home/luis/esp/esp-idf/components/esp_eth /home/luis/esp/esp-idf/components/esp_event /home/luis/esp/esp-idf/components/esp_gdbstub /home/luis/esp/esp-idf/components/esp_hid /home/luis/esp/esp-idf/components/esp_http_client /home/luis/esp/esp-idf/components/esp_http_server /home/luis/esp/esp-idf/components/esp_https_ota /home/luis/esp/esp-idf/components/esp_https_server /home/luis/esp/esp-idf/components/esp_hw_support /home/luis/esp/esp-idf/components/esp_lcd /home/luis/esp/esp-idf/components/esp_local_ctrl /home/luis/esp/esp-idf/components/esp_netif /home/luis/esp/esp-idf/components/esp_partition /home/luis/esp/esp-idf/components/esp_phy /home/luis/esp/esp-idf/components/esp_pm /home/luis/esp/esp-idf/components/esp_psram /home/luis/esp/esp-idf/components/esp_ringbuf /home/luis/esp/esp-idf/components/esp_rom /home/luis/esp/esp-idf/components/esp_system /home/luis/esp/esp-idf/components/esp_timer /home/luis/esp/esp-idf/components/esp_wifi /home/luis/esp/esp-idf/components/espcoredump /home/luis/Documents/PORYECT/uROS_mobile_robot_1/managed_components/espressif__bdc_motor /home/luis/Documents/PORYECT/uROS_mobile_robot_1/managed_components/espressif__mpu6050 /home/luis/esp/esp-idf/components/esptool_py /home/luis/esp/esp-idf/components/fatfs /home/luis/esp/esp-idf/components/freertos /home/luis/esp/esp-idf/components/hal /home/luis/esp/esp-idf/components/heap /home/luis/esp/esp-idf/components/http_parser /home/luis/esp/esp-idf/components/idf_test /home/luis/esp/esp-idf/components/ieee802154 /home/luis/esp/esp-idf/components/json /home/luis/esp/esp-idf/components/log /home/luis/esp/esp-idf/components/lwip /home/luis/Documents/PORYECT/uROS_mobile_robot_1/main /home/luis/esp/esp-idf/components/mbedtls /home/luis/Documents/PORYECT/uROS_mobile_robot_1/components/micro_ros_espidf_component /home/luis/esp/esp-idf/components/mqtt /home/luis/esp/esp-idf/components/newlib /home/luis/esp/esp-idf/components/nvs_flash /home/luis/esp/esp-idf/components/openthread /home/luis/esp/esp-idf/components/partition_table /home/luis/esp/esp-idf/components/perfmon /home/luis/esp/esp-idf/components/protobuf-c /home/luis/esp/esp-idf/components/protocomm /home/luis/esp/esp-idf/components/pthread /home/luis/esp/esp-idf/components/sdmmc /home/luis/esp/esp-idf/components/soc /home/luis/esp/esp-idf/components/spi_flash /home/luis/esp/esp-idf/components/spiffs /home/luis/esp/esp-idf/components/tcp_transport /home/luis/esp/esp-idf/components/ulp /home/luis/esp/esp-idf/components/unity /home/luis/esp/esp-idf/components/usb /home/luis/esp/esp-idf/components/vfs /home/luis/esp/esp-idf/components/wear_levelling /home/luis/esp/esp-idf/components/wifi_provisioning /home/luis/esp/esp-idf/components/wpa_supplicant /home/luis/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/luis/Documents/PORYECT/uROS_mobile_robot_1/build
[0/2] Re-checking globbed directories...
ninja: error: '../components/micro_ros_espidf_component/libmicroros.a', needed by 'micro_ros_esp_32.elf', missing and no known rule to make it

#### Additional information

the "-DUCLIENT_PROFILE_MULTITHREAD=ON" in colcon meta

Acuadros95, 2023-07-04:
Does the `[multithread_publisher](https://github.com/micro-ROS/micro_ros_espidf_component/tree/iron/examples/multithread_publisher)` example works for you?
Did you modify anything else?

>   1 package failed: microxrcedds_client

We will need the compilation errors of this package.

ALUIS97, 2023-07-04:
I tried the example and also I added in the colon meta of the micro ros component in a different test, also I tried to change the microxrcedds_client cmake and change multithread to on but I do not know how to compile the micro ros component without deleting the sub-packages of the element because the command idf.py clean-microros deleted all the sub packages like microxrcedds that is the big issue so if building the component with the colcon meta changed I don't know how to build the component with multithread in on state

Yams96, 2023-10-11:
I also cannot build the multithread_publisher example, but can build the int32_publisher example. Building in docker, docker file builds without error.
- micro_ros_espidf_component: iron
- esp-idf: release-v5.0(chosen by dockerfile)
- target: esp32