micro-ROS/micro_ros_espidf_component #202: addtwoints_server does not create node/service

Sebski123, 2023-07-08:
## Issue template

- Hardware description: ESP32 ([QuinLED-ESP32](https://quinled.info/quinled-esp32-ethernet/))
- RTOS: FreeRTOS
- Installation type: ethernet (LAN8720)
- Version or commit hash: humble

#### Steps to reproduce the issue
```
. $IDF_PATH/export.sh
cd examples/addtwoints_server
idf.py set-target esp32
idf.py menuconfig 
idf.py build
idf.py flash
```
then in a separate Ubuntu VM:
`docker run -it --rm --net=host microros/micro-ros-agent:iron udp4 --port 8888 -v6`

#### Expected behavior
microROS agent receives a connection from uC and creates ROS2 node/service

#### Actual behavior
microROS agent receives a connection from uC but does not create any nodes or services

**Output from agent**
```
osboxes@osboxes:~$ sudo docker run -it --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
[1688842062.190282] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1688842062.192982] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1688842068.598365] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5D 4B 2D EB 81 00 FC 01
[1688842068.599200] info     | Root.cpp           | create_client            | create                 | client_key: 0x5D4B2DEB, session_id: 0x81
[1688842068.599528] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x5D4B2DEB, address: 192.168.1.244:12790
[1688842068.600103] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5D4B2DEB, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1688842068.603285] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 56, data: 
0000: 81 80 00 00 01 07 2E 00 00 0A 00 01 01 03 00 00 20 00 00 00 00 01 FB 3F 18 00 00 00 61 64 64 5F
0020: 74 77 6F 69 6E 74 73 5F 63 6C 69 65 6E 74 5F 72 63 6C 63 00 03 00 00 00
[1688842068.605792] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.605906] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.611340] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.611365] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.611371] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.611376] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.611382] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.612949] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.620123] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.620268] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624714] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624818] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624827] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624833] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624838] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1688842068.624843] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x5D4B2DEB, len: 13, data: 
[... 174 lines cut ...]
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1688842068.693637] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x5D4B2DEB, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
```
**Output from `ros2 node list` and `ros2 service list`
[image]

#### Additional information
The examples were compiled in and flashed from Ubuntu 22.04 (WSL2).
microROS agent runs in Ubuntu 22.04 (VirtualBox).

The int32_publisher, multithread_publisher, and ping_pong examples all work as expected.

I have edited [/cnetwork_interfaces/uros_ethernet_netif.c](https://github.com/micro-ROS/micro_ros_espidf_component/blob/iron/network_interfaces/uros_ethernet_netif.c#L82) to include the following code so the ESP32 gets a static IP address:
```cpp
esp_netif_dhcpc_stop(eth_netif);
char* ip= "192.168.1.244";
char* gateway = "192.168.1.1";
char* netmask = "255.255.255.0";
esp_netif_ip_info_t info_t;
memset(&info_t, 0, sizeof(esp_netif_ip_info_t));
ipaddr_aton((const char *)ip, &info_t.ip.addr);
ipaddr_aton((const char *)gateway, &info_t.gw.addr);
ipaddr_aton((const char *)netmask, &info_t.netmask.addr);
esp_netif_set_ip_info(eth_netif, &info_t);```

Acuadros95, 2023-07-10:
That example modifies the default DOMAIN ID of the micro-ROS node:

```c
#define DOMAIN_ID 3
```

To visualize the node and the service, either modify this value to 0, or modify your ROS 2 domain ID to 3.

Sebski123, 2023-07-10:
That works, thanks

closed 2023-07-10 by Sebski123