micro-ROS/micro_ros_espidf_component #206: Micro-ros Ping utility issue

apurva-peppermint, 2023-08-09:
## Issue template

- Hardware description: esp32s3 devkit
- RTOS: freertos
- Installation type: standard
- Version or commit hash: humble

#### Steps to reproduce the issue
try the rmw_uros_ping_agent() utility to see whether the Agent is up or not it does not require creating a micro-ros context first... 

#### Expected behavior
When I turn the Agent on it should output as RMW_RET_OK else it should throw an error value

#### Actual behavior
[image]
There is a variable called agent_state_flag in the code which gives 0 when agent is off and 1 when agent is on...

[image]
Here I have not started the  agent yet

[image]
Here I have started the agent

[image]
Still it the rmw_uros_ping_agent() is not giving me a RMW_RET_OK even when the agent is alive

#### Additional information
[image]
The code snippet which I am using...

pablogs9, 2023-08-10:
Hello @apurva-peppermint please provide code for replicating this.

apurva-peppermint, 2023-08-10:
Hey @pablogs9 I'm using this code for the agent liveliness testing...

#include <string.h>
#include <stdio.h>
#include <unistd.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "driver/gpio.h"
#include "freertos/queue.h"
#include "driver/uart.h"

#include <uros_network_interfaces.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
// #include <std_msgs/msg/int32.h>
#include <std_msgs/msg/float32.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#ifdef CONFIG_MICRO_ROS_ESP_XRCE_DDS_MIDDLEWARE
#include <rmw_microros/rmw_microros.h>
#endif

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

int agent_state_flag = 0;

void agent_notifier(void *arg)
{
	while(1)
	{
		if(rmw_uros_ping_agent(100, 1) == RMW_RET_OK)
		{
			agent_state_flag = 1;
			printf("agent_state_flag = %d\n", agent_state_flag);
		}
		else
		{
			agent_state_flag = 0;
			printf("agent_state_flag = %d\n", agent_state_flag);
		}
	}		
}

void app_main(void)
{
#if defined(CONFIG_MICRO_ROS_ESP_NETIF_WLAN) || defined(CONFIG_MICRO_ROS_ESP_NETIF_ENET)
    ESP_ERROR_CHECK(uros_network_interface_initialize());
#endif

     xTaskCreate(agent_notifier,
     			"agent_notifier_task",
     			1024 * 4,
     			NULL,
     			CONFIG_MICRO_ROS_APP_TASK_PRIO ,
    			NULL);
}

pablogs9, 2023-08-10:
Please paste here the output of the micro-ROS Agent using the flag `-v6` to increase verbosity.

apurva-peppermint, 2023-08-10:
[image]

apurva-peppermint, 2023-08-10:
[image]

pablogs9, 2023-08-10:
If you use a normal micro-ROS example (publisher for example), does it works?

apurva-peppermint, 2023-08-10:
I have not tried that yet. So I want the client to start the context only when it detects that an agent has started or has become alive again. This ping function works when I first start the Agent and then my client. but if I try pinging the agent before creating anything as I have written in the code it does not work

pablogs9, 2023-08-10:
Probable dealing with an agent pinging on a non-initialized session can be weird due to the fact that it will try to open/close sockets...In your example, does the board send any UDP packet to multicast or to the agent IP address?

apurva-peppermint, 2023-08-10:
I will check and let you know here

apurva-peppermint, 2023-08-10:
hey @pablogs9 I just checked it and the board does not send any kind of UDP packet to the agents IP address. It only sends a packet when a micro ros session is on.