micro-ROS/micro_ros_espidf_component #207: Unable to build libmicroros.a correctly

lishuai-cau, 2023-10-04:
I use docker and esp idf v4.4 to make micro_ros humble branch, but it build failed.
I use the following steps：
  . $IDF_PATH/export.sh
  pip3 install catkin_pkg lark-parser empy colcon-common-extensions
  idf.py set-target esp32
  idf.py build
**The error message displayed is as follows：**
-- Adding linker script /opt/esp/idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Using Micro XRCE-DDS middleware
rm -f /project/esp/micro_ros_espidf_component/esp32_toolchain.cmake; \
cat /project/esp/micro_ros_espidf_component/esp32_toolchain.cmake.in | \
	sed "s/@CMAKE_C_COMPILER@/\/opt\/esp\/tools\/xtensa-esp32s3-elf\/esp-2021r2-patch5-8.4.0\/xtensa-esp32s3-elf\/bin\/xtensa-esp32s3-elf-gcc/g" | \
	sed "s/@CMAKE_CXX_COMPILER@/\/opt\/esp\/tools\/xtensa-esp32s3-elf\/esp-2021r2-patch5-8.4.0\/xtensa-esp32s3-elf\/bin\/xtensa-esp32s3-elf-g++/g" | \
	sed "s/@CFLAGS@/-mlongcalls  -ffunction-sections -fdata-sections/g" | \
	sed "s/@CXXFLAGS@/-mlongcalls  -ffunction-sections -fdata-sections/g" | \
	sed "s/@IDF_TARGET@/esp32s3/g" | \
	sed "s/@IDF_PATH@/\/opt\/esp\/idf/g" | \
	sed "s/@BUILD_CONFIG_DIR@/\/project\/esp\/micro_ros_espidf_component\/examples\/int32_publisher\/build\/config/g" \
	> /project/esp/micro_ros_espidf_component/esp32_toolchain.cmake
rm -rf micro_ros_dev; \
mkdir micro_ros_dev; cd micro_ros_dev; \
git clone -b humble https://github.com/ament/ament_cmake src/ament_cmake; \
git clone -b humble https://github.com/ament/ament_lint src/ament_lint; \
git clone -b humble https://github.com/ament/ament_package src/ament_package; \
git clone -b humble https://github.com/ament/googletest src/googletest; \
git clone -b humble https://github.com/ros2/ament_cmake_ros src/ament_cmake_ros; \
git clone -b humble https://github.com/ament/ament_index src/ament_index; \
colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=gcc;
rm -rf micro_ros_src; \
mkdir micro_ros_src; cd micro_ros_src; \
if [ "microxrcedds" = "embeddedrtps" ]; then \
	git clone -b humble https://github.com/micro-ROS/embeddedRTPS src/embeddedRTPS; \
	git clone -b humble https://github.com/micro-ROS/rmw_embeddedrtps src/rmw_embeddedrtps; \
else \
	git clone -b ros2 https://github.com/eProsima/Micro-XRCE-DDS-Client src/Micro-XRCE-DDS-Client; \
	git clone -b humble https://github.com/micro-ROS/rmw_microxrcedds src/rmw_microxrcedds; \
fi; \
git clone -b ros2 https://github.com/eProsima/micro-CDR src/micro-CDR; \
git clone -b humble https://github.com/micro-ROS/rcl src/rcl; \
git clone -b humble https://github.com/ros2/rclc src/rclc; \
git clone -b humble https://github.com/micro-ROS/rcutils src/rcutils; \
git clone -b humble https://github.com/micro-ROS/micro_ros_msgs src/micro_ros_msgs; \
git clone -b humble https://github.com/micro-ROS/rosidl_typesupport src/rosidl_typesupport; \
git clone -b humble https://github.com/micro-ROS/rosidl_typesupport_microxrcedds src/rosidl_typesupport_microxrcedds; \
git clone -b humble https://github.com/ros2/rosidl src/rosidl; \
git clone -b humble https://github.com/ros2/rmw src/rmw; \
git clone -b humble https://github.com/ros2/rcl_interfaces src/rcl_interfaces; \
git clone -b humble https://github.com/ros2/rosidl_defaults src/rosidl_defaults; \
git clone -b humble https://github.com/ros2/unique_identifier_msgs src/unique_identifier_msgs; \
git clone -b humble https://github.com/ros2/common_interfaces src/common_interfaces; \
git clone -b galactic https://github.com/ros2/example_interfaces src/example_interfaces; \
git clone -b humble https://github.com/ros2/test_interface_files src/test_interface_files; \
git clone -b humble https://github.com/ros2/rmw_implementation src/rmw_implementation; \
git clone -b humble https://github.com/ros2/rcl_logging src/rcl_logging; \
git clone -b humble https://gitlab.com/ros-tracing/ros2_tracing src/ros2_tracing; \
git clone -b humble https://github.com/micro-ROS/micro_ros_utilities src/micro_ros_utilities; \
    touch src/rosidl/rosidl_typesupport_introspection_cpp/COLCON_IGNORE; \
    touch src/rcl_logging/rcl_logging_log4cxx/COLCON_IGNORE; \
    touch src/rcl_logging/rcl_logging_spdlog/COLCON_IGNORE; \
    touch src/rclc/rclc_examples/COLCON_IGNORE; \
touch src/rcl/rcl_yaml_param_parser/COLCON_IGNORE; \
cp -rf /project/esp/micro_ros_espidf_component/extra_packages src/extra_packages || :; \
test -f src/extra_packages/extra_packages.repos && cd src/extra_packages && vcs import --input extra_packages.repos || :;
Cloning into 'src/ament_cmake'...
Cloning into 'src/Micro-XRCE-DDS-Client'...
Cloning into 'src/ament_lint'...
Cloning into 'src/rmw_microxrcedds'...
Cloning into 'src/ament_package'...
Cloning into 'src/micro-CDR'...
Cloning into 'src/googletest'...
[... 33 lines cut ...]
SyntaxError: invalid syntax
---
Failed   <<< ament_lint [0.19s, exited with code 1]
--- stderr: ament_cppcheck
Traceback (most recent call last):
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_ros/task/ament_python/build.py", line 51, in build
    setup_py_data = get_setup_data(self.context.pkg, env)
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_core/task/python/__init__.py", line 20, in get_setup_data
    return dict(pkg.metadata[key](env))
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_ros/package_augmentation/ros_ament_python.py", line 57, in getter
    return get_setup_information(
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
  File "/opt/esp/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 293, in _get_setup_information
    return ast.literal_eval(output)
  File "/usr/lib/python3.8/ast.py", line 59, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.8/ast.py", line 47, in parse
    return compile(source, filename, mode, flags,
  File "<unknown>", line 1
    {'package_data': {}, 'dist_files': [], 'src_root': None, 'dependency_links': [], 'setup_requires': [], 'eager_resources': None, 'entry_points': {'console_scripts': ['ament_cppcheck = ament_cppcheck.main:main']}, 'exclude_package_data': None, 'extras_require': {}, 'include_package_data': None, 'install_requires': ['setuptools'], 'namespace_packages': None, 'packages': ['ament_cppcheck'], 'python_requires': None, 'test_loader': None, 'test_runner': None, 'test_suite': None, 'tests_require': ['pytest'], 'use_2to3': None, 'zip_safe': True, 'cffi_modules': None, 'verbose': 1, 'dry_run': 0, 'help': 0, 'command_packages': None, 'script_name': 'setup.py', 'script_args': ['--dry-run'], 'command_options': {'install': {}}, 'package_dir': {}, 'py_modules': None, 'libraries': None, 'headers': None, 'ext_package': None, 'include_dirs': None, 'extra_path': None, 'scripts': None, 'data_files': [('share/ament_cppcheck', ['package.xml']), ('share/ament_index/resource_index/packages', ['resource/ament_cppcheck'])], 'password': '', 'command_obj': {}, 'have_run': {}, 'want_user_cfg': True, 'metadata': {'name': 'ament_cppcheck', 'version': '0.12.8', 'author': 'Dirk Thomas', 'author_email': 'dthomas@osrfoundation.org', 'maintainer': 'Michael Jeronimo, Michel Hidalgo', 'maintainer_email': 'michael.jeronimo@openrobotics.org, michel@ekumenlabs.com', 'url': 'https://github.com/ament/ament_lint', 'license': 'Apache License, Version 2.0', 'description': 'Static code analysis on C/C++ code using Cppcheck.', 'long_description': 'The ability to perform static code analysis on C/C++ code using Cppcheck\nand generate xUnit test result files.', 'keywords': ['ROS'], 'platforms': None, 'classifiers': ['Intended Audience :: Developers', 'License :: OSI Approved :: Apache Software License', 'Programming Language :: Python', 'Topic :: Software Development'], 'download_url': 'https://github.com/ament/ament_lint/releases', 'provides': None, 'requires': None, 'obsoletes': None, 'long_description_content_type': None, 'project_urls': {}, 'license_file': None, '_normalized_install_requires': {'setuptools': <Requirement('setuptools')>}, '_normalized_extras_require': {}}}

pablogs9, 2023-10-04:
This is a known issue, please check: https://github.com/colcon/colcon-python-setup-py/issues/56

lishuai-cau, 2023-10-04:
Thank you, I used this command and successfully solved my problem.

closed 2023-10-04 by lishuai-cau

ajeetjain94608, 2025-04-26:
> Thank you, I used this command and successfully solved my problem.

can you please share which command you used?