micro-ROS/micro_ros_espidf_component #211: microcdr C_STANDARD is set to invalid value '17'

Scoeerg, 2023-10-27:
# microcdr C_STANDARD is set to invalid value '17'
## Minimal Setup
I am using [Joy-It's NodeMCU ESP32](https://joy-it.net/en/products/SBC-NodeMCU-ESP32) via USB on an Ubuntu20.04 with installed, but not sourced ROS2 Foxy. 

Clean microros workspace, i.e.: 

```
cd /home/eric/Software/
git clone -b foxy https://github.com/micro-ROS/micro_ros_espidf_component.git
```

I have `ESP-IDF v5.0.4-dirty` installed in `home/eric/esp/esp-idf/`. I used the [VSCode ESP Extension](https://docs.espressif.com/projects/esp-idf/en/v4.2.3/esp32/get-started/vscode-setup.html). But for this minimal setup I 
will only use the terminal.

## Reproduce the Error

```
# open terminal without sourced ROS2 Foxy
. ~/esp/esp-idf/export.sh
```

Here's my log regarding `export.sh`:
[export.log](https://github.com/micro-ROS/micro_ros_espidf_component/files/13192328/export_sourcing.log)

```
idf.py --version
# ESP-IDF v5.0.4-dirty
cd /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher
```

then comes that part that goes wrong

```
idf.py set-target esp32
```

which results in 

```
...
Starting >>> microcdr
  opt = self.warn_dash_deprecation(opt, section)
--- stderr: microcdr
CMake Error at /home/eric/Software/micro_ros_espidf_component/micro_ros_src/build/microcdr/CMakeFiles/CMakeTmp/CMakeLists.txt:16 (add_library):
  C_STANDARD is set to invalid value '17'

CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:44 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  cmake/SuperBuild.cmake:19 (enable_language)
  CMakeLists.txt:54 (include)

---
Failed   <<< microcdr [0.50s, exited with code 1]
...
```

for the full log see [settarget.log](https://github.com/micro-ROS/micro_ros_espidf_component/files/13192550/settarget.log). CMake is later than `3.8`, see e.g. [this git issue](https://github.com/jmzkChain/jmzk/issues/33):

```
cmake --version
# cmake version 3.16.3
```

All other commands after this step fail, since some dependency is not built correctly.

Scoeerg, 2023-10-27:
It seems the built code resides in `$somepath/micro_ros_espidf_component/micro_ros_src/build`, I deleted this directory for a hopefully clean build and got the same error with shorter log, which I can share:

```
Directory '/home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
/home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults:2 CONFIG_ESP_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/eric/.espressif/python_env/idf5.0_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "int32_publisher" version: 1.0.0-1-g0d6a175
-- Adding linker script /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/eric/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
cd /home/eric/Software/micro_ros_espidf_component/micro_ros_src; \
unset AMENT_PREFIX_PATH; \
PATH="/home/eric/esp/esp-idf/components/espcoredump:/home/eric/esp/esp-idf/components/partition_table:/home/eric/esp/esp-idf/components/app_update:/home/eric/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/eric/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/eric/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/eric/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/eric/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/eric/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/eric/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/eric/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:/home/eric/.espressif/tools/ninja/1.10.2/:/home/eric/.espressif/python_env/idf5.0_py3.8_env/bin:/home/eric/esp/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"; \
. ../micro_ros_dev/install/local_setup.sh; \
colcon build \
	--merge-install \
[... cut ...] ome/eric/esp/esp-idf/components/ulp /home/eric/esp/esp-idf/components/unity /home/eric/esp/esp-idf/components/usb /home/eric/esp/esp-idf/components/vfs /home/eric/esp/esp-idf/components/wear_levelling /home/eric/esp/esp-idf/components/wifi_provisioning /home/eric/esp/esp-idf/components/wpa_supplicant /home/eric/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build
```

Scoeerg, 2023-10-28:
I believe is has something to do with 

https://github.com/micro-ROS/micro_ros_espidf_component/commit/a71fb638eb89d9ccf97b7142b29ec981986a0d44

this commit labeled "C_Standard workaround"

Scoeerg, 2023-10-28:
Since you determine which `C_Standard` to use based on ESP-IDF version within `CMakeLists.txt` I switched the version from `ESP-IDF v5.0.4-dirty` to `ESP-IDF v4.4.6-98-g5f257494c5`, yet again installed via VSCode Extension, which should change `C_Standard` from `17` to `11` to solve the problem. See:

```
# CMakeLists.txt
...
if (${IDF_VERSION_MAJOR} EQUAL 5)
    set(CMAKE_C_STANDARD 17)
else()
    set(CMAKE_C_STANDARD 11)
endif()
...
```

Long story short: does not work, but throws a different error:

```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/eric/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
-- Found Python3: /home/eric/.espressif/python_env/idf4.4_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "int32_publisher" version: 1.0.0-1-g0d6a175
-- Adding linker script /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/eric/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/eric/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
rm -f /home/eric/Software/micro_ros_espidf_component/esp32_toolchain.cmake; \
cat /home/eric/Software/micro_ros_espidf_component/esp32_toolchain.cmake.in | \
	sed "s/@CMAKE_C_COMPILER@/\/home\/eric\/.espressif\/tools\/xtensa-esp32-elf\/esp-2021r2-patch5-8.4.0\/xtensa-esp32-elf\/bin\/xtensa-esp32-elf-gcc/g" | \
	sed "s/@CMAKE_CXX_COMPILER@/\/home\/eric\/.espressif\/tools\/xtensa-esp32-elf\/esp-2021r2-patch5-8.4.0\/xtensa-esp32-elf\/bin\/xtensa-esp32-elf-g++/g" | \
	sed "s/@CFLAGS@/-mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections/g" | \
	sed "s/@CXXFLAGS@/-mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections/g" | \
	sed "s/@IDF_TARGET@/esp32/g" | \
	sed "s/@IDF_PATH@/\/home\/eric\/esp\/esp-idf/g" | \
	sed "s/@BUILD_CONFIG_DIR@/\/home\/eric\/Software\/micro_ros_espidf_component\/examples\/int32_publisher\/build\/config/g" \
	> /home/eric/Software/micro_ros_espidf_component/esp32_toolchain.cmake
rm -rf micro_ros_dev; \
mkdir micro_ros_dev; cd micro_ros_dev; \
git clone -b foxy https://github.com/ament/ament_cmake src/ament_cmake; \
git clone -b foxy https://github.com/ament/ament_lint src/ament_lint; \
[... 90 lines cut ...]
Cloning into 'src/tinydir_vendor'...

Summary: 7 packages finished [2.04s]
  1 package failed: ament_cmake_core
  2 packages aborted: ament_flake8 ament_pyflakes
  1 package had stderr output: ament_cmake_core
  47 packages not processed
make: *** [libmicroros.mk:38: /home/eric/Software/micro_ros_espidf_component/micro_ros_dev/install] Error 1
make: *** Waiting for unfinished jobs....
Cloning into 'src/rosidl'...
Cloning into 'src/rmw'...
Cloning into 'src/rcl_interfaces'...
[... cut ...] ome/eric/esp/esp-idf/components/ulp /home/eric/esp/esp-idf/components/unity /home/eric/esp/esp-idf/components/usb /home/eric/esp/esp-idf/components/vfs /home/eric/esp/esp-idf/components/wear_levelling /home/eric/esp/esp-idf/components/wifi_provisioning /home/eric/esp/esp-idf/components/wpa_supplicant /home/eric/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eric/Software/micro_ros_espidf_component/examples/int32_publisher/build
```

pablogs9, 2023-10-30:
micro-CDR sets C_STANDARD to 99:

https://github.com/eProsima/Micro-CDR/blob/3d1b17703c7cf4f22def2910bc845bdb5152d7b5/CMakeLists.txt#L114-L115

Scoeerg, 2023-10-30:
The initial Error regarding `C_Standard 17` takes place in

```
$(somePath)/micro_ros_espidf_component/micro_ros_src/build/microcdr/CMakeFiles/CMakeTmp/CMakeLists.txt:16
```

which to me seems to be an autogenerated `CMakeLists.txt` in a temporary directory. I would guess the default micro-CDR standard `99` will be overwritten by the command you are invoking within `idf.py set-target esp32`, namely:

```
colcon build \
	--merge-install \
	--packages-ignore-regex=.*_cpp \
	--metas /home/eric/Software/micro_ros_espidf_component/colcon.meta  \
	--cmake-args \
	"--no-warn-unused-cli" \
	-DCMAKE_POSITION_INDEPENDENT_CODE:BOOL=OFF \
	-DTHIRDPARTY=ON \
	-DBUILD_SHARED_LIBS=OFF \
	-DBUILD_TESTING=OFF \
	-DCMAKE_BUILD_TYPE=Release \
	-DCMAKE_TOOLCHAIN_FILE=/home/eric/Software/micro_ros_espidf_component/esp32_toolchain.cmake \
	-DCMAKE_VERBOSE_MAKEFILE=OFF \
        -DIDF_INCLUDES='-I/home/eric/.../network_interfaces' \
	-DCMAKE_C_STANDARD=17 \
	-DUCLIENT_C_STANDARD=17;
```

which might explain the temporary `CMakelists.txt` with **DCMAKE_C_STANDARD=17**. Thanks for your reply :)

pablogs9, 2023-10-30:
I wonder which is the difference between your environment and the passing CI of this repo https://github.com/micro-ROS/micro_ros_espidf_component/actions/workflows/nightly.yml

Please if you find a solution for this, is possible to open a PR?

Scoeerg, 2023-10-30:
Maybe I am blind, but:

Nightly builds **Ubuntu Latest** (points to Ubuntu 23.10?!) and **[humble, iron, rolling]** (rolling points to Iron again?!) 

where I described my setup to be **Ubuntu 20.04** and **Foxy**.

pablogs9, 2023-10-30:
>Nightly builds Ubuntu Latest (points to Ubuntu 23.10?!) 

Dockers runs on Github's Ubuntu latest, but the CI build is running on top of the official Espressif docker (`"espressif/idf:release-v4.3", "espressif/idf:release-v4.4", "espressif/idf:release-v5.0"`). So AFAIK, it is a "common" and "clean" environment: https://github.com/micro-ROS/micro_ros_espidf_component/blob/d0e63d9ab1d4b1b2777e68b715c5ea6d20ad68f4/.github/workflows/nightly.yml#L19

> and [humble, iron, rolling] (rolling points to Iron again?!)

Nightly runs from default branch with is `iron`, and there we are running in the 3 ROS 2 support distros: 

https://github.com/micro-ROS/micro_ros_espidf_component/blob/d0e63d9ab1d4b1b2777e68b715c5ea6d20ad68f4/.github/workflows/nightly.yml#L17

Where is Rolling pointing to Iron?

> where I described my setup to be Ubuntu 20.04 and Foxy. Including of course micro_ros foxy branch.

I'm not sure about the Ubuntu version compat wrt IDF, but micro-ROS Foxy is deprecated and EOL (as ROS 2 Foxy),

Scoeerg, 2023-10-30:
Github Ubuntu Latest is 22.04 see 

https://github.blog/changelog/2022-11-09-github-actions-ubuntu-latest-workflows-will-use-ubuntu-22-04/

And "Rolling" is meant to be Ros2 Rolling **not** the current (rolling)  version of Ros2. My bad. 

I am aware Foxy is EOL, but with some Nvidia Embedded Hardware currently only supports Ubuntu 20.04 based OS, hence I will have to stick to Foxy for now. 

Which micro_ros branch (or specific commit) should work, if it's not Foxy-Branch?

pablogs9, 2023-10-30:
`iron` shall work, but may be incompatibilities at type level with foxy

e271, 2023-11-03:
I tried it with C_Standard 11 and got the same problem as you, first with catkin and then with other depencies.

I solved with:
pip install catkin_pkg
pip install empy
pip install lark 

idf.py clean-microros
idf.py set-target esp32

Probably you made it already, but just in case, is also important to set the source so:
source /opt/ros/humble/setup.bash

Scoeerg, 2023-11-08:
Thanks to @e271 

**TL;DR** solves all problems:

```
# source ROS
source /opt/ros/foxy/setup.sh
# source ESP-IDF
cd esp-idf
source export.sh
idf.py --versions # returns ESP-IDF v4.4.6-150-g2bab3b36bc
# install missing dependencies in virtual environment:
pip install catkin_pkg
pip install empy
pip install lark
cd $yourpath//micro_ros_espidf_component/examples/int32_publisher
idf.py clean-microros
idf.py set-target esp32
```

**DETAILS**

**Problem 0)**

**One has to source ROS2** even though the [microros-esp README](https://github.com/micro-ROS/micro_ros_espidf_component#readme) says not to: _If you encounter issues during the build process, ensure that you are running in a clean shell environment without the ROS 2 setup script sourced._

@pablogs9 would be nice if you changed the README accordingly. 

**Problem 1)**
In order to compile with `C_Standard 11` and not `C_Standard 17`, **one must install ESP-IDF version 4** or earlier, see [my earlier comment in this issue](https://github.com/micro-ROS/micro_ros_espidf_component/issues/211#issuecomment-1783777346). 

E.g. via [VSCode Extension](https://docs.espressif.com/projects/esp-idf/en/v4.2.3/esp32/get-started/vscode-setup.html) or manually

```
git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf
cd esp-idf
sh ./install.sh
source export.sh
idf.py --versions # returns ESP-IDF v4.4.6-150-g2bab3b36bc
```
@pablogs9 A fix would be to change the 
```
# CMakeLists.txt
...
if (${IDF_VERSION_MAJOR} EQUAL 5)
    set(CMAKE_C_STANDARD 17)
else()
    set(CMAKE_C_STANDARD 11)
endif()
...
```

to always set the `CMAKE_C_STANDARD` to `11`. 

**Problem 2)**

Some dependencies are not automatically installed within the Python Virtual Environment, hence you must follow [@e271 's comment](https://github.com/micro-ROS/micro_ros_espidf_component/issues/211#issuecomment-1793227817).

closed 2023-11-08 by Scoeerg

pablogs9, 2023-11-08:
@Scoeerg it is stated in the Readme that you need to install dependecies: https://github.com/micro-ROS/micro_ros_espidf_component#dependencies

Also, the fact of not sourcing the ROS 2 environment has a reason: depending on the installation type, it would interfere with the micro-ROS packages build procedure.

With respect to C standard, it would be nice if you contribute back this patch in order improve this package.

Scoeerg, 2023-11-08:
@pablogs9 will do! Thanks for your support!