micro-ROS/micro_ros_espidf_component #212: Services getting random values from caller

flabrosse, 2023-11-27:
Caveat, I am not sure the espidf component is the issue.

- Hardware description: ESP32
- RTOS: FreeRTOS
- Installation type: esp-idf
- Version or commit hash: humble, galactic, latest micro-ros from updating from github (84f1ab2), as well as older hash (357379ecde).

In code I inherited [(here)](https://github.com/Aber-CRANC/cranc_microros_espidf/tree/galactic/components), we have services that run on the ESP, using our own service types but made of standard types (byte, uint8, bool, etc.).  When I call the services from a Linux computer (by hand using `ros2 service call` or from a program, the correct value is displayed in the message (on the command line), but the callback on the ESP receives a random value.

Maybe related, the service call never returns: I have to ctrl-C it from the command line.

pablogs9, 2023-11-27:
I cannot access the code you provided, could you provide a replicator?

flabrosse, 2023-11-27:
Grrrr, I had not noticed it was not accessible.  I'll try to come up with a minimum example.

flabrosse, 2023-11-28:
Example [here, expiring on 05/12/2022](https://www.transfernow.net/dl/20231128JPnnJUZU).  This contains an old (about 2 years old) version of micro-ros-espidf-component.  It won't compile with the latest (can't find rcl/rcl.h, as reported elsewhere).

pablogs9, 2023-11-28:
I won't use `rclc_executor_spin`, instead use the approach here: https://github.com/micro-ROS/micro_ros_espidf_component/blob/d0e63d9ab1d4b1b2777e68b715c5ea6d20ad68f4/examples/addtwoints_server/main/main.c#L74-L78

BTW, does the provided example work for you? https://github.com/micro-ROS/micro_ros_espidf_component/tree/iron/examples/addtwoints_server

flabrosse, 2023-11-28:
I can run addtwoints_server and client and they work, on my host computer (linux).  Interestingly enough, while the server is running, `ros2 service list` does not list the service, and I can't call on the CLI.  However, I do get the right values in the server, as sent by the client.  I now need to try this on an esp32.

flabrosse, 2023-12-22:
I am not sure exactly what the issue was, but it now works.  What I have changed is that I now run humble, while I was using galactic before.  I noticed that if I compiled the simple example and run it on the esp32 connected to my ubuntu 22.04 laptop running humble, it all worked, while   when running it on the RPI running galactic I had the problem.  I now updated the RPI to ubuntu 22.04 running humble and it all works.

However, I am still using the micro_ros_espdif_component from galactic, cloned [here](https://github.com/Aber-CRANC/micro_ros_espidf_component).  Indeed, if I try any of the branches from the official github, my code does not compile, missing rcl/rcl.h, as reported in issue #187.  I am still investigating this.

flabrosse, 2023-12-22:
Because my issue was a non-issue (maybe) but is solved in humble, I'll close it.

closed 2023-12-22 by flabrosse