micro-ROS/micro_ros_espidf_component #216: Can't create subscriber on ESP32-C3

2b-t, 2023-12-10:
#### Problem description
I have started recently to experiment with micro-ROS. While the examples involving just publishers such as [`int32_publisher`](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_publisher) are running fine, if I run any example involving a **subscriber** such as [`int32_sub_pub`](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_sub_pub) or [`ping_pong`](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/ping_pong) according to the official documentation using the provided **Dockerfile** this results in a **stack dump**.

- Hardware description: [Seeed Studio Xiao ESP32C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html)/[ESP32-C3-MINI-1](https://www.ebay.co.uk/itm/355174946337?var=624409881517)
- RTOS: FreeRTOS
- Installation type: `micro_ros_espidf_component`
- Version or commit hash: humble (27158f2369bd7bd02ce3cda61a785bf32ce10883)

#### Steps to reproduce the issue
In order to replicate this flash the ESP32-C3 board with one of the examples involving a subscriber in the following [`int32_sub_pub`](https://github.com/micro-ROS/micro_ros_espidf_component/tree/humble/examples/int32_sub_pub). In my case I open it in the container with the provided Dockerfile in interactive mode:
```bash
$ . $IDF_PATH/export.sh
$ cd examples/int32_sub_pub
$ idf.py set-target esp32c3
$ idf.py menuconfig
# Set micro-ROS agent IP (10.42.0.1) and WiFi credentials (test_hotspot, test_password)
$ idf.py build
$ sudo chmod o+rw /dev/ttyACM0
$ idf.py flash
```
then run the micro-ROS agent on the computer (`10.42.0.1`)
```bash
$ docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
```
and finally monitor the ESP32-C3 with:
```bash
$ idf.py monitor
```

#### Expected behavior
The ESP-C3 should publish its messages to the micro-ROS agent on the `int32_publisher` topic and wait from messages on the `int32_subscriber` topic. I should be able to see both topics on the micro-ROS agent side.

#### Actual behavior
The ESP32-C3 runs into a stack dump and reboots (For a full output see [subscriber_bug.txt](https://github.com/micro-ROS/micro_ros_espidf_component/files/13629494/subscriber_bug.txt)):
```bash
I (1599) main_task: Returned from app_main()
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x42015a2e  RA      : 0x4201596c  SP      : 0x3fcad780  GP      : 0x3fc91400  
0x42015a2e: rmw_wait at ??:?

0x4201596c: rmw_wait at ??:?

TP      : 0x3fc901b8  T0      : 0x4005890e  T1      : 0x0000000c  T2      : 0x00068081  
S0/FP   : 0x3fc94350  S1      : 0x3fcadafc  A0      : 0x3fca9a58  A1      : 0x00001000  
A2      : 0x00001000  A3      : 0x00000001  A4      : 0xffffffff  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xf4240000  S2      : 0x3fcadaf0  S3      : 0x00000064  
S4      : 0x3fcadb08  S5      : 0x3fcadb14  S6      : 0x05f5e100  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x3fcad8c4  S10     : 0x3fcadaec  S11     : 0x00000000  
T3      : 0x00000975  T4      : 0x00000000  T5      : 0x4200a2c8  T6      : 0x4200a2d0  
0x4200a2c8: __default_deallocate at allocator.c:?

0x4200a2d0: __default_allocate at allocator.c:?

MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0xd009f7d3  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x42015a2e in rmw_wait ()
#0  0x42015a2e in rmw_wait ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
ELF file SHA256: dc5c59390d980376
Rebooting...
```

#### Additional information
I have tried the following already to no avail:
- Remove the publisher, being left with an example with only a subscriber
- [Doubled the statically allocated memory for the ROS task](https://github.com/micro-ROS/micro_ros_setup/issues/526#issuecomment-1139360309)
- Added the following options to `colcon.meta` and proceed with a clean build, cleaning the workspace first with `$ idf.py clean-microros`:
  - Added [`-DRMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS=ON` to `microxrcedds_client.cmake-args`](https://github.com/micro-ROS/micro_ros_setup/issues/526#issuecomment-1139334773)
  - Increased the number of nodes `-DRMW_UXRCE_MAX_NODES`, publishers `-DRMW_UXRCE_MAX_PUBLISHERS`, subscribers `-DRMW_UXRCE_MAX_SUBSCRIPTIONS`, guard conditions `-DRMW_UXRCE_MAX_GUARD_CONDITION` under `rmw_microxrcedds.cmake-args`

If I uncomment the line that adds the subscription to the executor, e.g. `RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &recv_msg, &subscription_callback, ON_NEW_DATA));` the publisher runs just fine while clearly the subscriber will not work.

In input is highly appreciated!

pablogs9, 2023-12-11:
Which is the output of the micro-ROS Agent using the flag `-v6`?

2b-t, 2023-12-12:
_Hi @pablogs9_ 
Thanks for your quick reply!

The output of the micro-ROS agent with `-v6` is the following:

```bash
tobit@P500:~$ docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
[1702340879.676756] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1702340879.677083] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1702341006.162604] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 4F 60 5D E0 81 00 FC 01
[1702341006.162825] info     | Root.cpp           | create_client            | create                 | client_key: 0x4F605DE0, session_id: 0x81
[1702341006.162905] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x4F605DE0, address: 10.42.0.43:24558
[1702341006.163032] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4F605DE0, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1702341006.167917] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 64, data: 
0000: 81 80 00 00 01 07 36 00 00 0A 00 01 01 03 00 00 28 00 00 00 00 01 38 40 20 00 00 00 69 6E 74 33
0020: 32 5F 70 75 62 6C 69 73 68 65 72 5F 73 75 62 73 63 72 69 62 65 72 5F 72 63 6C 63 00 00 00 00 00
[1702341006.167946] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.169489] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.171055] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.172083] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.175093] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.175114] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.175150] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.175176] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.176210] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.177454] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.178572] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.179659] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.179680] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.181657] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.181681] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.183602] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.184753] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.184771] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.186676] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.186712] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1702341006.188713] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
[... cut ...]
[1702341006.248103] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 05 00 06 00 80
[1702341006.248167] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
[1702341006.250881] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4F605DE0, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
```

2b-t, 2023-12-12:
When checking out on a very old commit like f426aa27e29984d8e8fa307c5442a6466cd2fa69 (which happens to use `espressif/idf:release-v4.3` instead of `espressif/idf:release-v5.0`) it works without issues.

ghufran1942, 2024-10-03:
@2b-t Were you able to identify the issue? I am facing the same issue.

2b-t, 2024-10-03:
@ghufran1942 Unfortunately no, have not worked much with MicroROS ever since. As described above checking out to an earlier version worked for me.

kasperg3, 2024-12-13:
I am facing the same issue, also with the ESP32C3 from Seeeds. Did anyone find a solution using the latest version?

samiibnjamil, 2025-02-26:
same issue till now , anyone with a solution?, TIA