micro-ROS/micro_ros_espidf_component #217: An error is reported using the int32_publisher example “Failed status on line 54: 1. Aborting.”

jiafeng-1, 2023-12-15:
## used docker build 
I used mac os docker to compile the int32_publisher code for esp32c3. 
```bash
docker run -it --user espidf --volume="/etc/timezone:/etc/timezone:ro" -v  $(pwd):/micro_ros_espidf_component -v  /dev/cu.usbserial-110:/dev/cu.usbserial-110 --privileged --workdir /micro_ros_espidf_component
microros/esp-idf-microros:latest /bin/bash 
```
build+flash+monitor
```bash
cd examples/int32_publisher
idf.py set-target esp32c3
idf.py menuconfig
# build
idf.py build
# flash
idf.py --port 'rfc2217://host.docker.internal:4000?ign_set_control' flash
# monitor
idf.py --port 'rfc2217://host.docker.internal:4000?ign_set_control' monitor
```
log output
```bash
I (385) wifi:wifi driver task: 3fca20d8, prio:23, stack:6656, core=0
I (385) system_api: Base MAC address is not set
I (385) system_api: read default base MAC address from EFUSE
I (395) wifi:wifi firmware version: 17afb16
I (395) wifi:wifi certification version: v7.0
I (395) wifi:config NVS flash: enabled
I (405) wifi:config nano formating: disabled
I (405) wifi:Init data frame dynamic rx buffer num: 32
I (415) wifi:Init management frame dynamic rx buffer num: 32
I (415) wifi:Init management short buffer num: 32
I (425) wifi:Init dynamic tx buffer num: 32
I (425) wifi:Init static tx FG buffer num: 2
I (435) wifi:Init static rx buffer size: 1600
I (435) wifi:Init static rx buffer num: 10
I (435) wifi:Init dynamic rx buffer num: 32
I (445) wifi_init: rx ba win: 6
I (445) wifi_init: tcpip mbox: 32
I (445) wifi_init: udp mbox: 6
I (455) wifi_init: tcp mbox: 6
I (455) wifi_init: tcp tx win: 5744
I (465) wifi_init: tcp rx win: 5744
I (465) wifi_init: tcp mss: 1440
I (465) wifi_init: WiFi IRAM OP enabled
I (475) wifi_init: WiFi RX IRAM OP enabled
I (485) phy_init: phy_version 950,11a46e9,Oct 21 2022,08:56:12
I (515) wifi:mode : sta (7c:df:a1:61:bf:24)
I (515) wifi:enable tsf
I (515) wifi_station_netif: wifi_init_sta finished.
I (525) wifi:new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (1185) wifi:state: init -> auth (b0)
I (1215) wifi:state: auth -> assoc (0)
I (1215) wifi:state: assoc -> run (10)
I (1235) wifi:connected with TP-LINK_443A, aid = 7, channel 11, 40D, bssid = 48:5f:08:2b:44:3a
I (1235) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (1245) wifi:pm start, type: 1

I (1245) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (1295) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1415) wifi:<ba-add>idx:0 (ifx:0, 48:5f:08:2b:44:3a), tid:0, ssn:0, winSize:64
I (2255) esp_netif_handlers: sta ip: 192.168.1.110, mask: 255.255.255.0, gw: 192.168.1.3
I (2255) wifi_station_netif: got ip:192.168.1.110
I (2255) wifi_station_netif: connected to ap SSID:TP-LINK_443A password:jiafeng123
Failed status on line 54: 1. Aborting.
```

## used docker run micro_ros agent
docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6

log output
```bash

[1702658048.325821] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1702658048.326039] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

```
How should I fix it? Failed status on line 54: 1. Aborting.
```
![image](https://github.com/micro-ROS/micro_ros_espidf_component/assets/56590843/6ebc3bc0-5666-44a1-aee6-9a6c9eebcaae)

pablogs9, 2023-12-18:
Your board is not reaching the agent. Are you sure that the docker where you are running the micro-ROS Agent is accessible from the network?

jiafeng-1, 2023-12-18:
> 您的董事会尚未联系代理。您确定可以通过网络访问运行 micro-ROS Agent 的 docker 吗？
Thanks for the reply, with your reply I looked at the docker mapped port, re-mapped udp 8888 port and now it's ok
```bash
➜ docker run -it --rm -p 8888:8888/udp microros/micro-ros-agent:humble udp4 --port 8888 -v6
```

closed 2023-12-18 by jiafeng-1