micro-ROS/micro_ros_espidf_component #219: Size Limit on Sequence Type Members of Laser Scan Msg

xenn0n, 2023-12-31:
**Issue:**
I'm trying to publish a laser_scan_msg using microros publisher from esp32 to microros agent. I set the size of the ranges and and intensities into 360 which is a float type. However, when echoing the data recieved from the microros agent, the ranges and intensities only goes up to 128 members. 

**System Information:**
- Ubuntu 22.04.3 LTS
- esp32 dev board
- ros2 humble
- microros espidf component humble branch

I did this to allocate the memory:

```
#define BUF_LEN 360
static float mem_buf[BUF_LEN];

laserScan.ranges.capacity = BUF_LEN;
laserScan.ranges.data = mem_buf;
laserScan.ranges.size = 0;

laserScan.intensities.capacity = BUF_LEN;
laserScan.intensities.data = mem_buf;
laserScan.intensities.size = 0;
```

and publishing it with this dummy data:
```
for (size_t i = 0; i < BUF_LEN; i++) {
  laserScan.ranges.data[i] = i+1;
  laserScan.intensities.data[i] = i+1;

  laserScan.ranges.size++;
  laserScan.intensities.size++;
}
```

this is what it displayed when echoing the received data:
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: laser_scan
angle_min: -3.1241393089294434
angle_max: 3.1415927410125732
angle_increment: 0.01745329238474369
time_increment: 0.0
scan_time: 0.0
range_min: 0.15000000596046448
range_max: 12.0
ranges:
- 1.0
- 2.0
- 3.0
- 4.0
...... // I just shortened this part here
- 126.0
- 127.0
- 128.0
- '...'
intensities:
- 1.0
- 2.0
- 3.0
- 4.0
......
- 126.0
- 127.0
- 128.0
- '...'

```

I still have no idea where is the specific problem because as far as I know, the code compiles without a problem.

pablogs9, 2024-01-05:
use `-no-arr` option to see the actual size of the arrays

xenn0n, 2024-01-05:
sorry if this question may be a bit dumb, but how exactly do I do that?

xenn0n, 2024-01-06:
Hello this is the output when using `--no-arr` option:
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: laser_scan
angle_min: -3.1241393089294434
angle_max: 3.1415927410125732
angle_increment: 0.01745329238474369
time_increment: 0.0
scan_time: 0.0
range_min: 0.15000000596046448
range_max: 12.0
ranges: '<sequence type: float, length: 360>'
intensities: '<sequence type: float, length: 360>'
---
```
I'm confused as to why it only shows up to 128 members.

pablogs9, 2024-01-08:
I guess that the CLI limits the amount of data printed. Closing as not an issue.

closed 2024-01-08 by pablogs9