micro-ROS/micro_ros_espidf_component #220: app-colcon.meta vs colcon.meta

flabrosse, 2024-01-05:
This is not really an issue, but more to document what I found out after spending a lot of time having compilation errors.

A lot of the online documentation mentions the need for a file named `colcon.meta`, eg to use a different transport mechanism.  Not much of the documentation mentions a file named `app-colcon.meta`.

Renaming my `colcon.meta` to `app-colcon.meta` solved my compilation problem relating to `RMW_UXRCE_TRANSPORT_CUSTOM` not being defined.

pablogs9, 2024-01-08:
Both shall be used as you can see here:

https://github.com/micro-ROS/micro_ros_espidf_component/blob/dd5108214f747780ac401ab50a2b6ae97f1480fc/libmicroros.mk#L97

In that sense, `colcon.meta` is intended to configure the library, and `app-colcon.meta` is a sort of override for application-specific configurations. 

Make sure you clean the library completely before modifying it and changing colcon meta parameters.

flabrosse, 2024-01-08:
Yes, of course.  My point is that there is documentation out there that specifies colcon.meta in an application, not in the library.  I initially had a `colcon.meta`, inherited from a previous developer, that worked with the galactic version.  When I changed to humble, it was complaining about the transport not being configured.  Changing the file to `app-colcon.meta` fixed it.

Yes, systematically was doing clean-microros, clean and fullclean.

pablogs9, 2024-01-08:
AFAIK the only documentation that this package has is its own README.
And you are right, `app-colcon.meta` is not documented there.

Contributions are welcomed, please feel free to open a PR to include a paragraph mentioning it.

closed 2025-07-15 by 4ntn