micro-ROS/micro_ros_espidf_component #221: Adding a timeout parameter to rclc_support_init()

flabrosse, 2024-01-05:
To avoid waiting for a fixed amount of time at the start of my code on the ESP32, waiting for the micro-ros agent to start on my Raspberry Pi, I use `rclc_support_init()` and `rclc_support_fini()` in a loop until the connection is established.  At the same time I flash an LED to indicate we are trying to connect.  However, `rclc_support_init()` takes a long time to timeout if the agent is not available, producing a very slow flashing LED.

Is it possible to set a timeout for `rclc_support_init()`?

pablogs9, 2024-01-08:
There is a recommended way of achieving your goals, is using ping functionality, please check `rmw_uros_ping_agent` at: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html#state-machine-approach

flabrosse, 2024-01-08:
I seem to remember this only works for the UDP transport (I think I read this in an issue here).  But maybe this has changed.  I'll have a look.

flabrosse, 2024-01-10:
Just to confirm and close this issue.  The pinging as linked above works even using the serial transport, and is a neat solution.  Thank you!

closed 2024-01-10 by flabrosse