micro-ROS/micro_ros_espidf_component #226: idf.py build failing in Ubuntu 20.04

Pankaj-s-dev, 2024-01-13:
I am following the steps but i am getting this error `cp: cannot stat 'librcutils.a': No such file or directory`

But with the same steps i am able to build it in Ubuntu 22.04. i even tried in different machine running Ubuntu 20.04, there also i am getting the same error.

I am trying to build **foxy**.

below are the logs that got while building :- 

```
16:54 $ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
/home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults:2 CONFIG_ESP_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
Compiler supported targets: xtensa-esp32s2-elf

-- App "int32_publisher" version: 1.0.0-1-g0d6a175
-- Adding linker script /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_system/ld/esp32s2/sections.ld.in
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.api.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.libgcc.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-data.ld
-- Adding linker script /home/williams/esp/esp-idf/components/esp_rom/esp32s2/ld/esp32s2.rom.spiflash.ld
-- Adding linker script /home/williams/esp/esp-idf/components/soc/esp32s2/ld/esp32s2.peripherals.ld
echo /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/atomic_workaround; \
	mkdir /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/atomic_workaround; cd /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/atomic_workaround; \
	/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar x /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/lib/librcutils.a; \
	/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_fetch_add_8; \
	if [ 5 -ge 4 ] && [ 1 -ge 3 ]; then \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_load_8; \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_store_8; \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_exchange_8; \
	fi; \
	if [ 5 -ge 4 ] && [ 1 -ge 4 ]; then \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_load_8; \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_store_8; \
	fi; \
	if [ 5 -ge 5 ] && [ 1 -ge 0 ]; then \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_load_8; \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_store_8; \
		/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip atomic_64bits.c.obj --strip-symbol=__atomic_exchange_8; \
	fi; \
	/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar rc -s librcutils.a *.obj; \
	cp -rf librcutils.a  /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/lib/librcutils.a; \
	rm -rf /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/atomic_workaround; \
	cd ..;
/home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/atomic_workaround
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar: /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/lib/librcutils.a: No such file or directory
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip: 'atomic_64bits.c.obj': No such file
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip: 'atomic_64bits.c.obj': No such file
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip: 'atomic_64bits.c.obj': No such file
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-strip: 'atomic_64bits.c.obj': No such file
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar: *.obj: No such file or directory
cp: cannot stat 'librcutils.a': No such file or directory
mkdir -p /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
	mkdir -p $folder; cd $folder; /home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar x $file; \
	for f in *; do \
		mv $f ../$folder-$f; \
	done; \
	cd ..; rm -rf $folder; \
done ; \
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/williams/dev_ws/src/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/include /home/williams/dev_ws/src/micro_ros_espidf_component/include;
/home/williams/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-ar: *.obj: No such file or directory
cp: cannot stat 'libmicroros.a': No such file or directory
cp: cannot stat '/home/williams/dev_ws/src/micro_ros_espidf_component/micro_ros_src/install/include': No such file or directory
make: *** [libmicroros.mk:127: /home/williams/dev_ws/src/micro_ros_espidf_component/libmicroros.a] Error 1
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls micro_ros_espidf_component mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/williams/esp/esp-idf/components/app_trace /home/williams/esp/esp-idf/components/app_update /home/williams/esp/esp-idf/components/bootloader /home/williams/esp/esp-idf/components/bootloader_support /home/williams/esp/esp-idf/components/bt /home/williams/esp/esp-idf/components/cmock /home/williams/esp/esp-idf/components/console /home/williams/esp/esp-idf/components/cxx /home/williams/esp/esp-idf/components/driver /home/williams/esp/esp-idf/components/efuse  [... cut ...]
-- Build files have been written to: /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/build
ninja: error: '/home/williams/dev_ws/src/micro_ros_espidf_component/libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it
ninja failed with exit code 1, output of the command is in the /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stderr_output_15625 and /home/williams/dev_ws/src/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stdout_output_15625
```

pablogs9, 2024-01-15:
Did you install the dependencies? https://github.com/micro-ROS/micro_ros_espidf_component?tab=readme-ov-file#dependencies

Pankaj-s-dev, 2024-01-19:
> Did you install the dependencies? https://github.com/micro-ROS/micro_ros_espidf_component?tab=readme-ov-file#dependencies

Hi @pablogs9,

Sorry for late reply, i had installed the dependencies. 

Right now i am using docker for the building and flashing the firmware.

closed 2024-03-27 by Pankaj-s-dev