micro-ROS/micro_ros_espidf_component #227: Cannot build int32_publisher_custom_transport with docker container

sweidling, 2024-01-23:
Hi all,

I am trying to build the example "int32_publisher_custom_transport" with the docker container using this command:
```
docker run -it --rm --user espidf --volume="/etc/timS transporezone:/etc/timezone:ro" -v  ${pwd}:/micro_ros_espidf_component -v  /dev:/dev --privileged --workdir /micro_ros_espidf_component microros/esp-idf-microros:latest /bin/bash  -c "cd examples/int32_publisher_custom_transport; idf.py menuconfig build"
```
It results in this error message:
```
/micro_ros_espidf_component/examples/int32_publisher_custom_transport/main/main.c: In function 'app_main':    
/micro_ros_espidf_component/examples/int32_publisher_custom_transport/main/main.c:97:2: error: #error micro-ROS transports misconfigured
   97 | #error micro-ROS transports misconfigured
```

I guess it has something to do with the app-colcon.meta which overrides the colcon.meta, right? I also set DRMW_UXRCE_TRANSPORT=custom in the colcon.meta. Same result.

Did I miss an important step?

pablogs9, 2024-01-23:
Does it work in a common IDF environment? Is it related only to the docker?

sweidling, 2024-01-23:
> Does it work in a common IDF environment? Is it related only to the docker?

Sorry forgot to mention. I had the same error with ESP-IDF environment without docker.

pablogs9, 2024-01-23:
It is working on CI: https://github.com/micro-ROS/micro_ros_espidf_component/actions/runs/7620914872/job/20756440364

Could you check which steps are you missing from this procedure: https://github.com/micro-ROS/micro_ros_espidf_component/blob/e41a9d882011219d89631ef688fe502aa9df7956/.github/workflows/nightly.yml#L70-L79

sweidling, 2024-01-23:
Thank you. I will check the steps. My guess is that I have to do a clean before changing the colcon.meta.

sweidling, 2024-01-23:
It was the missing clean before changing the colcon.meta and rebuild. Problem solved. Thank you for your support!

closed 2024-01-23 by sweidling