micro-ROS/micro_ros_espidf_component #233: vscode+espidf5.2 Could not complete

wang-yanguo, 2024-03-01:
Hardware : ESP32
RTOS:FreeRTOS
demo: micro_ros_espidf_component\examples\int32_publisher
Development environment : windows +VSCode +ESP IDF5.2
[image]

wang-yanguo, 2024-03-01:
-- Found Git: C:/Users/ns/.espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/ns/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/ns/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/ns/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/ESP32/ros/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file C:/ESP32/ros/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
C:/ESP32/ros/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults:2 CONFIG_ESP_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT
-- Compiler supported targets: xtensa-esp-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/ns/.espressif/python_env/idf5.2_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "int32_publisher" version: 4.1.0-5-g2b541f8
-- Adding linker script C:/ESP32/ros/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/ns/esp/v5.2/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
CMake Error at C:/ESP32/ros/micro_ros_espidf_component/CMakeLists.txt:77 (message):
  FAILED: The system cannot find the specified file。

-- Configuring incomplete, errors occurred!
See also "C:/ESP32/ros/micro_ros_espidf_component/examples/int32_publisher/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\ESP32\ros\micro_ros_espidf_component\examples\int32_publisher\build -S c:\ESP32\ros\micro_ros_espidf_component\examples\int32_publisher" terminated with exit code: 1.

Hailaylin, 2024-03-09:
Windows 不行，别试了，换Ubuntu22.04环境重新装ESP-IDFv5.2框架，直接运行example，下面是成功运行的样例，别死磕，用VMware17 pro创建个虚拟机，直通串口到Ubuntu就好了。希望能帮到你
[image]

wang-yanguo, 2024-03-11:
确实这样，已放弃window下开发，linux可以直接编译

muecknerm, 2025-03-07:
does the microros realy not work under windows with esp-idf?