micro-ROS/micro_ros_espidf_component #235: idf.py menuconfig error related to xtensa-esp-elf tool

gsoto-tellez, 2024-03-19:
I am using a Jetson Xavier running ROS2 and an ESP32 WROOME
I installed the ESP-IDF and created my own freertos project and pasted the micro-ros component folder. 

[image]

When attempting to select the middleware as instructed using idf.py menuconfig generated an error:

[image]

It is concerning a libmicroros.a file which is not in the microros component folder. 

Am I supposed to add this file manually?

pablogs9, 2024-03-19:
Could you first build the library and then try to open the menuconfig? Let me know if this works

gsoto-tellez, 2024-03-20:
Yes I built it and got these errors:
[image]

pablogs9, 2024-03-20:
Why are you using micro_ros_setup? This is an standalone module for ESPIDF.

gsoto-tellez, 2024-03-20:
My apologies I sent the wrong screenshot. That was when I was following the micro-ROS documentation. This is the error I get when I build. 
[image]

pablogs9, 2024-03-21:
I need the complete log and the steps you followed to reach this error.

gsoto-tellez, 2024-03-22:
I created my own freertos project and pasted the microros component folder in it. Then I added the dependencies following this:
[image]
I ran idf.py then attempted to select the micro-ROS middleware following this: 
[image]

The complete error from running idf.py menuconfig is:
objpkjnfp@teambot-desktop:~/freertos_project$ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/objpkjnfp/freertos_project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/objpkjnfp/.espressif/python_env/idf5.3_py3.8_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/objpkjnfp/freertos_project"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/home/objpkjnfp/freertos_project/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/objpkjnfp/freertos_project/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /home/objpkjnfp/.espressif/python_env/idf5.3_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "my-freertos-project" version: 1
-- Adding linker script /home/objpkjnfp/freertos_project/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/objpkjnfp/freertos_project/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/objpkjnfp/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
	mkdir -p $folder; cd $folder; /home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-ar x $file; \
	for f in *; do \
		mv $f ../$folder-$f; \
	done; \
	cd ..; rm -rf $folder; \
done ; \
/home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/micro_ros_src/install/include /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/include;
/home/objpkjnfp/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp-elf-ar: *.obj: No such file or directory
cp: cannot stat 'libmicroros.a': No such file or directory
cp: cannot stat '/home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/micro_ros_src/install/include': No such file or directory
make: *** [libmicroros.mk:139: /home/objpkjnfp/freertos_project/components/micro_ros_espidf_component/libmicroros.a] Error 1
CMake Error at components/micro_ros_espidf_component/CMakeLists.txt:77 (message):
  FAILED: 2

-- Configuring incomplete, errors occurred!
See also "/home/objpkjnfp/freertos_project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

mackaygrange, 2024-10-20:
I am having this exact same issue. Is there any update on this?

Ericc99, 2025-02-12:
I have encontered the same issue as well, any updates?

k-jayanth, 2025-02-15:
Facing the same problem here also on humble branch

rehmaan-dev, 2025-02-20:
[image]

I'd like to point out that if you;re using any additional pckg, you should check that the addiotnal or extra pckg is being rebuilt when building the firmware, if you look in the extra-pckg folder, you may get the idea and besides ensure your build completes sucessfully, if not try building for esp32 dev kit board and see if it builds.

k-jayanth, 2025-02-26:
yes it got resolved after cloning and rebuilding from scratch. Thanks

bernacbs, 2025-04-21:
Hello, in 2024 I faced this issue and somehow managed to resolve it. This week, I encountered it again and found what is supposedly the solution:

Based on my tests, try changing your Linux locale. Apparently, it has some relation to the error that occurred. In all the tests I performed, this fixed the libmicroros compilation issue.

`sudo apt update && sudo apt install locales`
`sudo locale-gen en_US en_US.UTF-8`
`sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8`
`export LANG=en_US.UTF-8`