micro-ROS/micro_ros_espidf_component #236: Can't create freertos tasks

SynKolbasyn, 2024-03-23:
## Issue template

- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: iron

#### Steps to reproduce the issue
1. I created the microros_ws/firmware/freeertos_apps/apps/test_project/ folder
2. I have created two files in this folder:
app-colcon.meta:
```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=0",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
            ]
        }
    }
}
```

main.c:
```
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>

#include <rclc/rclc.h>
#include <rclc/executor.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t publisher1;
rcl_publisher_t publisher2;

void task_1(void* param);
void task_2(void* param);

void appMain(void* arg) {
  // support & allocator
  rcl_allocator_t allocator = rcl_get_default_allocator();
  rclc_support_t support;
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // node
  rcl_node_t node;
  RCCHECK(rclc_node_init_default(&node, "esp32", "", &support));

  // publishers
  RCCHECK(rclc_publisher_init_default(&publisher1, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "/pub1"));
  RCCHECK(rclc_publisher_init_default(&publisher2, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "/pub2"));

  // tasks
  xTaskCreate(task_1, "task_1", 2048, NULL, 1, NULL);
  xTaskCreate(task_2, "task_2", 2048, NULL, 1, NULL);

  while (1) {
    vTaskDelay(1000000 / portTICK_PERIOD_MS);
  }
[... 113 lines cut ...]
[1711220922.696608] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0102790A, len: 36, data:
0000: 81 80 06 00 01 07 1C 00 00 10 00 15 05 03 00 00 0E 00 00 00 00 12 01 00 03 00 01 3F 0A 00 00 00
0020: 00 00 00 13
[1711220922.697019] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0102790A, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1711220922.697103] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0102790A, len: 14, data:
0000: 81 80 06 00 05 01 06 00 00 10 00 15 00 00
[1711220922.697123] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0102790A, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
[1711220922.733249] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0102790A, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 07 00 00 00 80
[1711220922.733306] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0102790A, len: 16, data:
0000: 81 80 07 00 07 01 08 00 00 12 00 05 00 00 00 00
[1711220922.733414] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 4, data:
0000: 00 00 00 00
[1711220922.733499] debug    | SerialAgentLinux.cpp | send_message             | [** <<SER>> **]        | client_key: 0x0102790A, len: 13, data:
0000: 81 00 00 00 0A 01 05 00 08 00 00 00 80
[1711220923.147581] debug    | SerialAgentLinux.cpp | recv_message             | [==>> SER <<==]        | client_key: 0x0102790A, len: 24, data:
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 06 DA 84 66 81 00 FC 01
[1711220923.149737] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x0102790A
[1711220923.149774] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x0102790A, address: 0
```
and messages in topics are constantly = 0

#### Additional information
Examples from the microros_ws/firmware/freertos_apps/apps/ folder run without problems

pablogs9, 2024-04-01:
You are not using [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component):

```
ros2 run micro_ros_setup configure_firmware.sh test_app -t serial
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh
source install/local_setup.bash
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v 6
```

SynKolbasyn, 2024-04-01:
Oh my God, what a fool I am. I apologize for this misunderstanding.

closed 2025-07-15 by 4ntn