micro-ROS/micro_ros_espidf_component #237: microxrcedds_client fails to build with -DUCLIENT_PROFILE_MULTITHREAD=ON

danielsmc, 2024-03-30:
## Issue template

- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type: idf 5.2 and the Docker image
- Version or commit hash: humble

#### Steps to reproduce the issue
Clone this repo and run the docker image with the command in the readme (but build example/multithread_publisher rather than int32_publisher.)

#### Expected behavior
The example project successfully builds.

#### Actual behavior
The build fails with the following output:
```
<trimmed output from before it started to fail-- I can post more if helpful>
--- stderr: microxrcedds_client
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/input_best_effort_stream.h:27,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_best_effort_stream_internal.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_best_effort_stream.c:1:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/input_reliable_stream.h:28,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/./input_reliable_stream_internal.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/input_reliable_stream.c:4:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/output_best_effort_stream.h:27,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_best_effort_stream_internal.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_best_effort_stream.c:1:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/output_reliable_stream.h:28,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_reliable_stream_internal.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/output_reliable_stream.c:1:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:76: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_best_effort_stream.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:90: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/input_reliable_stream.c.obj] Error 1
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:104: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_best_effort_stream.c.obj] Error 1
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:118: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/output_reliable_stream.c.obj] Error 1
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/output_best_effort_stream.h:27,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/stream_storage.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/stream_storage_internal.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/stream/stream_storage.c:1:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:132: CMakeFiles/microxrcedds_client.dir/src/c/core/session/stream/stream_storage.c.obj] Error 1
In file included from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/output_best_effort_stream.h:27,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/stream/stream_storage.h:23,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/core/session/session.h:29,
                 from /micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/src/c/core/session/session.c:1:
/micro_ros_espidf_component/micro_ros_src/src/Micro-XRCE-DDS-Client/include/uxr/client/profile/multithread/multithread.h:33:10: fatal error: FreeRTOS.h: No such file or directory
   33 | #include "FreeRTOS.h"
      |          ^~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:174: CMakeFiles/microxrcedds_client.dir/src/c/core/session/session.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:83: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:136: all] Error 2
---
Failed   <<< microxrcedds_client [0.64s, exited with code 2]
Aborted  <<< rosidl_adapter [2.22s]
Aborted  <<< rosidl_runtime_c [0.80s]
Aborted  <<< tracetools [0.99s]

Summary: 5 packages finished [5.61s]
  1 package failed: microxrcedds_client
  3 packages aborted: rosidl_adapter rosidl_runtime_c tracetools
  4 packages had stderr output: microxrcedds_client rcutils rosidl_cli rosidl_runtime_c
  56 packages not processed
make: *** [libmicroros.mk:87: /micro_ros_espidf_component/micro_ros_src/install] Error 2
CMake Error at /micro_ros_espidf_component/CMakeLists.txt:74 (message):
  FAILED: 2

-- Configuring incomplete, errors occurred!
See also "/micro_ros_espidf_component/examples/multithread_publisher/build/CMakeFiles/CMakeOutput.log".
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
cmake failed with exit code 1, output of the command is in the /micro_ros_espidf_component/examples/multithread_publisher/build/log/idf_py_stderr_output_146 and /micro_ros_espidf_component/examples/multithread_publisher/build/log/idf_py_stdout_output_146```

#### Additional information

pablogs9, 2024-04-01:
Working on this here: https://github.com/micro-ROS/micro_ros_espidf_component/pull/238

pablogs9, 2024-04-01:
Merged https://github.com/micro-ROS/micro_ros_espidf_component/pull/238, could you recheck if it works now?

danielsmc, 2024-04-04:
It works! Thank you for your work on this project. It's very useful.