micro-ROS/micro_ros_espidf_component #242: Adding a custom interface or message to esp-idf microROS

Ncouch64, 2024-04-22:
Hello, another basic question again because I have gotten a little confused about how to get a custom service working. I read other issues about adding the ROS2 built package to an extra_packages directory. When I add in and try and build however I see the error output
 `
CMake Error at CMakeLists.txt:9 (find_package):
  By not providing "Findcv_interface.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_interface", but CMake did not find one.

  Could not find a package configuration file provided by "cv_interface" with
  any of the following names:

    cv_interfaceConfig.cmake
    cv_interface-config.cmake

  Add the installation prefix of "cv_interface" to CMAKE_PREFIX_PATH or set
  "cv_interface_DIR" to a directory containing one of the above files.  If
  "cv_interface" provides a separate development package or SDK, be sure it
  has been installed.
`
I may be skipping a step here as I am still really new to the microROS system and am working my way through various tutorials. Thanks for the assistance!

Ncouch64, 2024-04-30:
I have figured this out.

closed 2024-04-30 by Ncouch64

Roboprotos, 2025-07-07:
Hi @Ncouch64 , I'm trying to integrate a custom message on micro-ROS for an ESP32S3. What steps did you follow, is there a documentation for that ?

Ncouch64, 2025-07-07:
Hello @Roboprotos, its been some time since I worked with microROS. I was following the docs here: https://micro.ros.org/docs/tutorials/advanced/create_new_type/.