micro-ROS/micro_ros_espidf_component #245: How can I configure Security QoS?

YiJunKo, 2024-05-27:
## Issue template

- Hardware description: ESP32-S3-EYE
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I would like to know how to set up my security QoS on micro-ROS (that is, on the XRCE-DDS client). Since there is currently no manual available, I am unsure how to implement it.

pablogs9, 2024-05-28:
Hello @YiJunKo, micro-ROS does not support DDS Security.

closed 2024-05-28 by pablogs9

YiJunKo, 2024-05-28:
Hello @pablogs9  , thanks your answers!
I would also like to ask where I can find the list of QoS supported by micro-ROS？

YiJunKo, 2024-05-28:
Hi @pablogs9 , 
I have referred to the following: https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/ and also looked at the colcon.meta of this project. If I am developing and flashing using this project, are the only available QoS options as follows?
"rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=2",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1"
            ]
        },

pablogs9, 2024-05-28:
Those are not QoS, those are middleware configuration parameters, please take a look at micro-ROS documentation: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html