micro-ROS/micro_ros_espidf_component #248: Agent does not communicate with Client

badmanwillis, 2024-07-10:
## Agent does not communicate with Client

- Hardware description: ESP32-WROOM-32D
- RTOS: FreeRTOS 
- Installation type: esp-idf extension V5.2.2, micro_ros_espidf_component
- Version or commit hash: Iron
- Application: int32_publisher

### Steps to reproduce the issue

#### Install Ubuntu 22.04LTS, ROS2 IRON, and Visual Studio Code
Don't set a Domain ID, this may confuse micro_ros (i don't fully understand why yet) https://docs.ros.org/en/iron/Concepts/Intermediate/About-Domain-ID.html

#### Install esp-idf extension via VScode 
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
Confirm installation with a "hello world" tutorial https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html#get-started-linux-macos-first-steps

#### Install the micro_ros_espidf_component 
https://github.com/micro-ROS/micro_ros_espidf_component/tree/iron
The github states "You can clone this repo directly in the components folder of your project." This can be interpreted two ways. I've tried both locations with the same end result...
- /home/warhorse/esp/v5.2.2/esp-idf/components
- /home/warhorse/uros_esp/firmware/toolchain/esp-idf/components
`. $HOME/esp/v5.2.2/esp-idf/export.sh`
`pip3 install catkin_pkg lark-parser colcon-common-extensions`
`git clone https://github.com/micro-ROS/micro_ros_espidf_component.git –branch iron`

#### Create ESP32 Client
- I have been able to use micro_ros_setup to create/configure/build/flash firmware, having worked through numerous issues, before learning via github issues that the micro_ros_espidf_component is the dev preferred method. 
- For reproducing this issue, I'm using the micro_ros_espidf_component installed within the vsCode extension installation
`. $HOME/esp/v5.2.2/esp-idf/export.sh`
`cd examples/int32_publisher`
##### Set target board 
`idf.py set-target esp32`
`idf.py menuconfig`
##### Set your micro-ROS configuration and WiFi credentials under micro-ROS Settings
##### Set the SSID & Password
`idf.py build`
##### find your usb port using
`ls /dev/tty*`
##### **Hold the boot button** on the ESP32 device, and flash the code.
`idf.py -p /dev/ttyUSB0 flash`
`idf.py -p /dev/ttyUSB0 monitor`

#### Create Agent
- While I have been able to build an agent using micro_ros_setup, my understanding is that micro_ros_espidf_component is the successor to micro_ros_setup, although that then raises a new question of how an end user is to create their own agent, without docker or micro_ros_setup...
- For the sake of reproduction, I'm using the Docker command

`sudo docker run -it --rm --net=host microros/micro-ros-agent:iron udp4 --port 8888 -v6`
It may be that the docker agent can't reach the network, try
`sudo docker run -it --rm -p 8888:8888/udp microros/micro-ros-agent:iron udp4 --port 8888 -v6`

#### Expected behavior
The agent and client communicate, publishing data to a ros2 topic.

#### Actual behavior
[1720633694.195512] info 	| UDPv4AgentLinux.cpp | init                 	| running...         	| port: 8888
[1720633694.195862] info 	| Root.cpp       	| set_verbose_level    	| logger setup       	| verbose_level: 6

No matter what i try, I cannot seem to get the agent to: create client, establish session etc. I was only able to achieve this using micro_ros_setup for the pin_pong demo https://micro.ros.org/docs/tutorials/core/first_application_linux/ 

#### Additional information
I intend to contribute documentation/tutorials for the setup of esp32 devices, once I have this working. If an end user starts their uros journey by visiting the website, they are liable to be mislead (as I was) into following tutorials that are not applicable to ESP32 devices in the year 2024.
https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/ 
https://medium.com/@SameerT009/connect-esp32-to-ros2-foxy-5f06e0cc64df 
https://robofoundry.medium.com/esp32-micro-ros-actually-working-over-wifi-and-udp-transport-519a8ad52f65
It is not made apparent that the esp-idf-component is the way to go, until one starts reading through github issues

maartin0, 2024-07-11:
I can confirm I've got the same issue with a ESP-WROOM-32 using `iron`. Although it's erroring and I get the following output from `idf.py monitor`:
```
I (1835) esp_netif_handlers: sta ip: 192.168.137.5, mask: 255.255.255.0, gw: 192.168.137.1
I (1835) wifi_station_netif: got ip:192.168.137.5
I (1835) wifi_station_netif: connected to ap SSID:<redacted> password:<redacted>
I (3665) wifi:<ba-add>idx:0 (ifx:0, <redacted>), tid:0, ssn:5, winSize:64
Failed status on line 54: 1. Aborting.
```
Where line 54 is the `rclc_support_init_with_options` call:
```c
53: // create init_options
54: RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));
```
I haven't tried putting debugging in the `rclc_support_init_with_options` method yet, but I'll say if I figure out what the issue is

badmanwillis, 2024-07-11:
@maartin0 I'm also getting this error when attempting `idf.py monitor` for the micro_ros_espidf_component int32_publisher.

However, it does work for me when attempting the [esp-idf get-started "hello world" example](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html#step-5-first-steps-on-esp-idf) tutorial. 

Would you mind trying to reproduce this? It may help us narrow down the problem.

maartin0, 2024-07-11:
Dang sorry you're right it was a network issue, I had tried 2 different networks so assumed one of them would allow UDP traffic - I ended up testing it by running an AP on a raspberry pi (which was also acting as the agent). Thanks for fixing my issue although sorry I couldn't help more!

badmanwillis, 2024-07-11:
@maartin0  please could you give some further explanation as to your problem and what steps you took to test & confirm the working solution. It may help me decipher my problem.

maartin0, 2024-07-11:
Yeah sure, I just kept on reconfiguring the network until [this](https://serverfault.com/a/733921) neat way of checking if UDP traffic went through actually worked between my raspberry pi and my laptop (since I guessed if my laptop could send UDP traffic to the pi then the ESP would be able to as well):
> 1. Install netcat `sudo apt install nc`
> 2. on server listen UDP port: nc -ul 6111 (add the -6 option if you're testing an ipv6 connection)
> 3. on client nc -u <server> 6111
> 4. type anything on client and hit enter - you should see this text on server

If you have control over your network I'd try and get that to work, but if you want to set up a raspberry pi AP I followed [this](https://raspberrypi-guide.github.io/networking/create-wireless-access-point) tutorial although it's a little bit more involved than explained if you're running Ubuntu since you have to disable the `netplan` and `systemd-resolved` services as well
I would recommend just following [this](https://www.tomshardware.com/how-to/raspberry-pi-access-point) tutorial since it's so much easier assuming you've installed the default Raspbian

trabelsim, 2024-07-18:
I was dealing with the same problem and found out that some "known rules" are not mentioned in the repo description.
My environment:

- ROS 2 Jazzy
- Ubuntu 24.04

I fixed my problem by:

1. Setting MICRO_ROS_AGENT_IP to the desktop IP:

- idf.py menuconfig -> micro-ROS Settings -> micro-ROS Agent IP = DESKTOP_IP_HERE

2. Setting the firewall rule to don't block the UDP traffic on port 8888:

-  `sudo ufw allow 8888/udp`

Once that's done, build the app, flash it, activate the monitoring and run as well the ROS agent (as described in the readme).
Hope that helped someone speed up the setup.

badmanwillis, 2024-07-28:
@trabelsim Thank you kindly, setting MICRO_ROS_AGENT_IP to the desktop IP, and adding the firewall rule solved the problem.

closed 2024-07-28 by badmanwillis