micro-ROS/micro_ros_espidf_component #256: Readme example not compiling esp32s3

tomy983, 2024-09-19:
## Issue template

- Hardware description: esp32s3
- RTOS: FreeRTOS
- Installation type: as per example
- Version or commit hash: humble

#### Steps to reproduce the issue
follow the readme

#### Expected behavior
compiles
#### Actual behavior
does not compile after the idf.py build command
#### Additional information
```idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/build
Executing "ninja all"...
[1/2] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig
Loading defaults file /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults...
/home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/sdkconfig.defaults:2 CONFIG_ESP_TASK_WDT was replaced with CONFIG_ESP_TASK_WDT_INIT 
-- Compiler supported targets: xtensa-esp-elf
-- App "int32_publisher" version: 5.0.1-1-g002cbc1-dirty
-- Adding linker script /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/pi/esp/v5.3.1/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Using Micro XRCE-DDS middleware
mkdir -p /home/pi/bno086/components/micro_ros_espidf_component/micro_ros_src/libmicroros; cd /home/pi/bno086/components/micro_ros_espidf_component/micro_ros_src/libmicroros; \
for file in $(find /home/pi/bno086/components/micro_ros_espidf_component/micro_ros_src/install/lib/ -name '*.a'); do \
	folder=$(echo $file | sed -E "s/(.+)\/(.+).a/\2/"); \
	mkdir -p $folder; cd $folder; /home/pi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-ar x $file; \
	for f in *; do \
		mv $f ../$folder-$f; \
	done; \
	cd ..; rm -rf $folder; \
done ; \
/home/pi/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-ar rc -s libmicroros.a *.obj; cp libmicroros.a /home/pi/bno086/components/micro_ros_espidf_component; \
cd ..; rm -rf libmicroros; \
cp -R /home/pi/bno086/components/micro_ros_espidf_component/micro_ros_src/install/include /home/pi/bno086/components/micro_ros_espidf_component/include;
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_vfs_console esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls micro_ros_espidf_component mqtt newlib nvs_flash nvs_sec_provider openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport touch_element ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/pi/esp/v5.3.1/esp-idf/components/app_trace /home/pi/esp/v5.3.1/esp-idf/components/app_update /home/pi/esp/v5.3.1/esp-idf/components/bootloader /home/pi/esp/v5.3.1/esp-idf/components/bootloader_support /home/pi/esp/v5.3.1/esp-idf/components/bt /home/pi/esp/v5.3.1/esp-idf/components/cmock /home/pi/esp/v5.3.1/esp-idf/components/console /home/pi/esp/v5.3.1/esp-idf/components/cxx /home/pi/esp/v5.3.1/esp-idf/components/driver /home/pi/esp/v5.3.1/esp-idf/components/efuse /home/pi/esp/v5.3.1/esp-idf/components/esp-tls /home/pi/esp/v5.3.1/esp-idf/components/esp_adc /home/pi/esp/v5.3.1/esp-idf/components/esp_app_format /home/pi/esp/v5.3.1/esp-idf/components/esp_bootloader_format /home/pi/esp/v5.3.1/esp-idf/components/esp_coex /home/pi/esp/v5.3.1/esp-idf/components/esp_common /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_ana_cmpr /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_cam /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_dac /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_gpio /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_gptimer /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_i2c /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_i2s /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_isp /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_jpeg /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_ledc /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_mcpwm /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_parlio /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_pcnt /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_ppa /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_rmt /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_sdio /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_sdm /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_sdmmc /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_sdspi /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_spi /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_touch_sens /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_tsens /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_uart /home/pi/esp/v5.3.1/esp-idf/components/esp_driver_usb_serial_jtag /home/pi/esp/v5.3.1/esp-idf/components/esp_eth /home/pi/esp/v5.3.1/esp-idf/components/esp_event /home/pi/esp/v5.3.1/esp-idf/components/esp_gdbstub /home/pi/esp/v5.3.1/esp-idf/components/esp_hid /home/pi/esp/v5.3.1/esp-idf/components/esp_http_client /home/pi/esp/v5.3.1/esp-idf/components/esp_http_server /home/pi/esp/v5.3.1/esp-idf/components/esp_https_ota /home/pi/esp/v5.3.1/esp-idf/components/esp_https_server /home/pi/esp/v5.3.1/esp-idf/components/esp_hw_support /home/pi/esp/v5.3.1/esp-idf/components/esp_lcd /home/pi/esp/v5.3.1/esp-idf/components/esp_local_ctrl /home/pi/esp/v5.3.1/esp-idf/components/esp_mm /home/pi/esp/v5.3.1/esp-idf/components/esp_netif /home/pi/esp/v5.3.1/esp-idf/components/esp_netif_stack /home/pi/esp/v5.3.1/esp-idf/components/esp_partition /home/pi/esp/v5.3.1/esp-idf/components/esp_phy /home/pi/esp/v5.3.1/esp-idf/components/esp_pm /home/pi/esp/v5.3.1/esp-idf/components/esp_psram /home/pi/esp/v5.3.1/esp-idf/components/esp_ringbuf /home/pi/esp/v5.3.1/esp-idf/components/esp_rom /home/pi/esp/v5.3.1/esp-idf/components/esp_system /home/pi/esp/v5.3.1/esp-idf/components/esp_timer /home/pi/esp/v5.3.1/esp-idf/components/esp_vfs_console /home/pi/esp/v5.3.1/esp-idf/components/esp_wifi /home/pi/esp/v5.3.1/esp-idf/components/espcoredump /home/pi/esp/v5.3.1/esp-idf/components/esptool_py /home/pi/esp/v5.3.1/esp-idf/components/fatfs /home/pi/esp/v5.3.1/esp-idf/components/freertos /home/pi/esp/v5.3.1/esp-idf/components/hal /home/pi/esp/v5.3.1/esp-idf/components/heap /home/pi/esp/v5.3.1/esp-idf/components/http_parser /home/pi/esp/v5.3.1/esp-idf/components/idf_test /home/pi/esp/v5.3.1/esp-idf/components/ieee802154 /home/pi/esp/v5.3.1/esp-idf/components/json /home/pi/esp/v5.3.1/esp-idf/components/log /home/pi/esp/v5.3.1/esp-idf/components/lwip /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/main /home/pi/esp/v5.3.1/esp-idf/components/mbedtls /home/pi/bno086/components/micro_ros_espidf_component /home/pi/esp/v5.3.1/esp-idf/components/mqtt /home/pi/esp/v5.3.1/esp-idf/components/ne [... cut ...]
/home/pi/esp/v5.3.1/esp-idf/components/newlib/stdatomic.c:77: multiple definition of `__atomic_store_8'; /home/pi/bno086/components/micro_ros_espidf_component/libmicroros.a(librcutils-atomic_64bits.c.obj):atomic_64bits.c:(.text.__atomic_store_8+0x0): first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stderr_output_312161 and /home/pi/bno086/components/micro_ros_espidf_component/examples/int32_publisher/build/log/idf_py_stdout_output_312161
```

aleroldan156, 2024-09-19:
try this
https://bit.ly/4gwC82r
If you don't have the c compliator, install it.(gcc or clang)

tomy983, 2024-09-19:
Dangerous File Blocked

The file you attempted to download was determined to be dangerous. For your protection, MediaFire does not enable distribution of dangerous files.

Could you describe the fix in the shared content?

closed 2024-09-19 by tomy983