micro-ROS/micro_ros_espidf_component #268: Compilation of micro-ROS fails inside Docker image due to version of empy

2b-t, 2024-11-10:
I ran into a **problem with the [empy](https://pypi.org/project/empy/) version** (4.2) that is installed in the [**Docker image**](https://github.com/micro-ROS/micro_ros_espidf_component/blob/jazzy/docker/Dockerfile) that comes with this project (`AttributeError: module 'em' has no attribute 'BUFFERED_OPT'` when executing `$ idf.py set-target esp32s3`).

- Hardware description: Seeed Studio Xiao ESP32-S3 Sense as well as ESP32-C3; bug should apply to all micro-controllers alike
- RTOS: FreeRTOS; should apply to all RTOS alike
- Installation type: micro_ros_setup
- Version or commit hash: Tested in 27df26a965fd50aaa1a10bfb73a377503acf39c6 (latest humble), c2d581ae1d39f0b23eb472306c021df5cd468f22 (latest Jazzy) as well as f426aa27e29984d8e8fa307c5442a6466cd2fa69; probably applies to every Docker image that is built from scratch

#### Steps to reproduce the issue
Recompile the Docker image from its [Dockerfile](https://github.com/micro-ROS/micro_ros_espidf_component/blob/jazzy/docker/Dockerfile) and recompile the micro-ROS library, e.g. by compiling one of the examples such as:
```
$ export LC_ALL=C
$ source /opt/esp/idf/export.sh
$ idf.py clean-microros
$ cd examples/int32_publisher
$ idf.py set-target esp32s3
```

#### Expected behavior
The compilation of micro-ROS succeeds.

#### Actual behavior
The compilation fails with the following error message:
```
--- stderr: builtin_interfaces
CMake Warning at /micro_ros_espidf_component/micro_ros_src/install/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'rcutils' exports library 'dl' which couldn't be found
Call Stack (most recent call first):
  /micro_ros_espidf_component/micro_ros_src/install/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_runtime_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_runtime_c/cmake/rosidl_runtime_cConfig.cmake:41 (include)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_typesupport_introspection_c/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_typesupport_introspection_c/cmake/rosidl_typesupport_introspection_cConfig.cmake:41 (include)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:13 (find_package)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)
CMake Error at /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(/opt/esp/python_env/idf4.3_py3.6_env/bin/python3 -m
  rosidl_adapter --package-name builtin_interfaces --arguments-file
  /micro_ros_espidf_component/micro_ros_src/build/builtin_interfaces/rosidl_adapter__arguments__builtin_interfaces.json
  --output-dir
  /micro_ros_espidf_component/micro_ros_src/build/builtin_interfaces/rosidl_adapter/builtin_interfaces
  --output-file
  /micro_ros_espidf_component/micro_ros_src/build/builtin_interfaces/rosidl_adapter/builtin_interfaces.idls)
  returned error code 1:
  AttributeError processing template 'msg.idl.em'
  Traceback (most recent call last):
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/resource/__init__.py", line 51, in evaluate_template
      em.BUFFERED_OPT: True,
  AttributeError: module 'em' has no attribute 'BUFFERED_OPT'
  During handling of the above exception, another exception occurred:
  Traceback (most recent call last):
    File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/main.py", line 55, in main
      pathlib.Path(relative_path), output_dir)
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/__init__.py", line 20, in convert_to_idl
      package_dir, package_name, interface_file, output_dir / 'msg')
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/micro_ros_espidf_component/micro_ros_src/install/lib/python3.6/site-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()
  AttributeError: 'NoneType' object has no attribute 'shutdown'
Call Stack (most recent call first):
  /micro_ros_espidf_component/micro_ros_src/install/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
---
Failed   <<< builtin_interfaces [2.46s, exited with code 1]
```

#### Additional information
This is an error commonly encountered with the ROS build system [related to a change in empy](https://stackoverflow.com/questions/77642155/attributeerror-module-object-has-no-attribute-raw-opt/77656642#77656642). Inside the container empy seems to be installed as version 4.2 system-wide as well as inside a virtual environment. As a **temporary** fix I uninstalled both of them as follows and **re-installed a compatible version of empy**:

```bash
$ pip3 uninstall empy
$ pip3 install empy==3.3.4
# The environment will be called differently depending on the version of the esp-idf Docker base image
$ source /opt/esp/python_env/idf5.0_py3.8_env/bin/activate
$ sudo chown -R $(whoami) /opt/esp/python_env/idf5.0_py3.8_env
$ pip3 uninstall empy
$ pip3 install empy==3.3.4
```
After cleaning micro-ROS with `$ idf.py clean-microros` the compilation just works fine.

pablogs9, 2024-11-11:
Could you try installing `empy==3.3.4`? Like here:

https://github.com/micro-ROS/micro_ros_espidf_component/blob/c2d581ae1d39f0b23eb472306c021df5cd468f22/.github/workflows/ci.yml#L33

2b-t, 2024-11-17:
_Thanks for the quick reply @pablogs9,_
Just https://github.com/micro-ROS/micro_ros_espidf_component/blob/c2d581ae1d39f0b23eb472306c021df5cd468f22/.github/workflows/ci.yml#L33 does not work and first sourcing esp-idf and then forcing the downgrade of empy https://github.com/micro-ROS/micro_ros_espidf_component/blob/c2d581ae1d39f0b23eb472306c021df5cd468f22/.github/workflows/ci.yml#L32-L33 does not work either and fails with the following error message:

```
Installing collected packages: empy
  Attempting uninstall: empy
    Found existing installation: empy 4.2
    Uninstalling empy-4.2:
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp/python_env/idf5.0_py3.8_env/bin/__pycache__/em.cpython-38.pyc'
Check the permissions.
```
The user that is created inside the container does not have the required permissions to reinstall empy.
What works is adding it to https://github.com/micro-ROS/micro_ros_espidf_component/blob/c2d581ae1d39f0b23eb472306c021df5cd468f22/docker/install_micro_ros_deps_script.sh#L8

pablogs9, 2024-11-18:
https://github.com/micro-ROS/micro_ros_espidf_component/pull/273 merged, closing. Thanks for the PR

closed 2024-11-18 by pablogs9