micro-ROS/micro_ros_espidf_component #281: Failed status on line 54: 1. Aborting while running int32_publisher example in esp32s3

DhamuPradeep, 2025-02-28:
## Running the int32_publisher  example in esp32s3 

To reproduce the issue:
```
cd ~

mkdir microros

cd microros
```
Source the micro-ROS component for the esp32s3 platform.

```
docker run -it --rm --name micro_ros --volume="/etc/timezone:/etc/timezone:ro" -v  $(pwd):/micro_ros -v  /dev:/dev --privileged --workdir /micro_ros microros/esp-idf-microros:latest /bin/bash
```

```
git clone https://github.com/micro-ROS/micro_ros_espidf_component.git

pip3 install catkin_pkg lark-parser empy colcon-common-extensions
```

```
cd examples/int32_publisher

idf.py set-target esp32s3
```
Use menuconfig to insert the WiFi Credentials and specific settings.

```
idf.py menuconfig

idf.py build

idf.py flash

idf.py monitor
```

Start a micro-ROS agent container to connect to your newly set-up esp node.

```
docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
```

I can able to flash the program into the esp32s3 program.
Below is the log i am getting while running  **idf.py monitor**

```
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Running idf_monitor in directory /micro_ros/micro_ros_espidf_component/examples/int32_publisher
Executing "/opt/esp/python_env/idf5.0_py3.8_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /micro_ros/micro_ros_espidf_component/examples/int32_publisher/build/int32_publisher.elf -m '/opt/esp/python_env/idf5.0_py3.8_env/bin/python' '/opt/esp/idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048df9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1654
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
I (24) boot: ESP-IDF v5.0-541-g885e501d99 2nd stage bootloader
I (25) boot: compile time 04:42:08
I (25) boot: chip revision: v0.2
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1eff4h (126964) map
I (116) esp_image: segment 1: paddr=0002f01c vaddr=3fc96500 size=00ffch (  4092) load
I (117) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=85d48h (548168) map
I (220) esp_image: segment 3: paddr=000b5d70 vaddr=3fc974fc size=02fa4h ( 12196) load
I (223) esp_image: segment 4: paddr=000b8d1c vaddr=40374000 size=12460h ( 74848) load
I (250) boot: Loaded app from partition at offset 0x10000
I (250) boot: Disabling RNG early entropy source...
I (262) cpu_start: Pro cpu up.
I (262) cpu_start: Starting app cpu, entry point is 0x40375364
0x40375364: call_start_cpu1 at /opt/esp/idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (276) cpu_start: Pro cpu start user code
I (276) cpu_start: cpu freq: 160000000 Hz
I (276) cpu_start: Application information:
I (279) cpu_start: Project name:     int32_publisher
I (285) cpu_start: App version:      5.0.1-8-g031fac2-dirty
I (291) cpu_start: Compile time:     Feb 28 2025 04:42:06
I (297) cpu_start: ELF file SHA256:  ca7d9c28e4190e51...
I (303) cpu_start: ESP-IDF:          v5.0-541-g885e501d99
I (309) cpu_start: Min chip rev:     v0.0
I (314) cpu_start: Max chip rev:     v0.99 
I (319) cpu_start: Chip rev:         v0.2
I (324) heap_init: Initializing. RAM available for dynamic allocation:
I (331) heap_init: At 3FCA0B20 len 00048BF0 (290 KiB): D/IRAM
I (337) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (344) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (350) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (357) spi_flash: detected chip: gd
I (361) spi_flash: flash io: dio
W (364) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (378) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (440) wifi_station_netif: ESP_WIFI_MODE_STA
I (440) pp: pp rom version: e7ae62f
I (440) net80211: net80211 rom version: e7ae62f
I (450) wifi:wifi driver task: 3fcf4760, prio:23, stack:6656, core=0
I (450) system_api: Base MAC address is not set
I (450) system_api: read default base MAC address from EFUSE
I (470) wifi:wifi firmware version: 17afb16
I (470) wifi:wifi certification version: v7.0
I (470) wifi:config NVS flash: enabled
I (470) wifi:config nano formating: disabled
I (470) wifi:Init data frame dynamic rx buffer num: 32
I (480) wifi:Init management frame dynamic rx buffer num: 32
I (480) wifi:Init management short buffer num: 32
I (490) wifi:Init dynamic tx buffer num: 32
I (490) wifi:Init static tx FG buffer num: 2
I (500) wifi:Init static rx buffer size: 1600
I (500) wifi:Init static rx buffer num: 10
I (500) wifi:Init dynamic rx buffer num: 32
I (510) wifi_init: rx ba win: 6
I (510) wifi_init: tcpip mbox: 32
I (520) wifi_init: udp mbox: 6
I (520) wifi_init: tcp mbox: 6
I (520) wifi_init: tcp tx win: 5744
I (530) wifi_init: tcp rx win: 5744
I (530) wifi_init: tcp mss: 1440
I (540) wifi_init: WiFi IRAM OP enabled
I (540) wifi_init: WiFi RX IRAM OP enabled
I (600) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (640) wifi:mode : sta (94:a9:90:08:94:68)
I (640) wifi:enable tsf
I (640) wifi_station_netif: wifi_init_sta finished.
I (1250) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2230) wifi:state: init -> auth (b0)
I (2250) wifi:state: auth -> assoc (0)
I (2260) wifi:state: assoc -> run (10)
I (2270) wifi:connected with MCW_CBE, aid = 2, channel 6, BW20, bssid = 68:28:cf:4d:92:20
I (2270) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (2280) wifi:pm start, type: 1

I (2280) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (2280) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2300) wifi:<ba-add>idx:0 (ifx:0, 68:28:cf:4d:92:20), tid:0, ssn:1, winSize:64
I (3290) esp_netif_handlers: sta ip: 192.168.63.40, mask: 255.255.255.0, gw: 192.168.63.1
I (3290) wifi_station_netif: got ip:192.168.63.40
I (3290) wifi_station_netif: connected to ap SSID:xxxx password:xxxx
Failed status on line 54: 1. Aborting.

```

Waelhechmi, 2025-03-01:
hi i want to create a new project not build a existing example

SerNikLar, 2025-08-11:
Hi, did you find a solution to the problem? I have the same problem esp32s3 and in that line of code 54.