micro-ROS/micro_ros_espidf_component #290: Debugging RCL_RET_ERROR during init

hierophect, 2025-05-18:
### Setup

- Hardware description: ESP32-S3 on Devkit C
- RTOS: FreeRTOS/IDF
- Installation type: integrating component with Circuitpython
- Version or commit hash: foxy

### Issue

I'm recieving an error code of 1 (`RCL_RET_ERROR`) when initializing microros (running `rclc_support_init()` with a custom allocator), and I am trying to understand my options for debugging it. I've found this to be difficult because:

1. RCL_RET_ERROR is generic and does not provide hints at the error origin
2. Microros does not support RCL logging with `rcutils`
3. Because the component both fetches and compiles the ROS middleware during builds it's difficult to inject ESP_LOGI or other print commands directly into the middleware
4. There's no documented process for debug builds and GDB on espressif is limited and struggles to access component memory correctly.

I'm hoping someone might be able to provide advice on how to approach this problem. Is it better to consider patching the middleware to include custom logging output, or to continue hammering at GDB, which has been giving incorrect variable reports? Advice from anyone with experience with vague runtime errors returned from the ROS middleware without the benefit of `rcutils` would be greatly appreciated.

4ntn, 2025-05-19:
Hello, @hierophect,

My suggestion is that you fork the rclc to add your own logging, and then modify `libmicroros.mk` to clone from your repository and branch.

Also, foxy is EOL since June 2023. If foxy is not mandatory for your setup, I'd suggest migrating to humble or jazzy.

hierophect, 2025-05-30:
Closing this as @4ntn's suggestion works and was ultimately the best solution I tried. I'd also like to explore the use of `rmw_uros_error_handling` as an alternative, but haven't had the chance to try it yet.

closed 2025-05-30 by hierophect